sn4k3/UVtools #837: [Bug] UVTools No longer working on Windows 11 After 4.2.1 to 4.2.2 Upgrade

sdmorrey, 2024-02-19:
### System

```Text
Everything was working great. Then I upgraded to 4.2.2 which appeared to go well and literally nothing happened. I don't even see an error message. It just fails to launch.

Thinking it was something weird with my setup I uninstalled 4.2.2 and reinstalled 4.2.1 which of course worked. Then I tried to install 4.2.2 from the distribution .msi and the same thing happened.

I'm unsure if it's a problem with upgrading or with 4.2.2 itself, but I've tried every way I can think of to upgrade this.
```

### Printer and Slicer

Not relevant

### Description of the bug

It does not launch at all, there is no crash screen and I'm not sure where it would log to. 
I've even tried launching from the command prompt but I get nada.

### How to reproduce

Install 4.2.1 using winget as per the front page
Upgrade to 4.2.2
Try to launch
Tear hair in frustration

sn4k3, 2024-02-19:
For Windows users v4.2.1->v4.2.2:
For who is upgrading from v4.2.1 to v4.2.2 need to uninstall UVtools first and then install the new version. Or upgrade, and run installer again and select repair action.
Due to the downgrade of the UI framework it wont be upgraded via MSI installer, that causes the program to not launch.

sdmorrey, 2024-02-20:
That's great information to know. Thank you so much! Perhaps the readme could be updated?