sn4k3/UVtools #841: Can't import layers right after exporting them[Bug]

yaroslaw, 2024-02-24:
### System

```Text
UVtools v4.2.2 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Core(TM) i5-3470 CPU @ 3.20GHz
Processor cores: 4
Memory RAM: 10.03 / 15.88 GB
Runtime: win10-x64
Framework: .NET 6.0.27
AvaloniaUI: 11.0.7
OpenCV: 4.8.1

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1040    UA: 1920 x 1040

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: m4_nut_m4_thumb_nut1.pws [Version: 1] [Class: PhotonWorkshopFile]
```

### Printer and Slicer

- Printer: Anycubic Photon S
- Slicer: Lychee

### Description of the bug

I'm exporting layers to images, moving model on a plate and run a script processing layers, then importing images. 
It gives an error 
"Could not import and fit 1 out of 1 format(s) (Insufficient or unequal build space)."
No matter of what import method I use or number of images. 

Layer resolution: w=1440, h=2560

exported image size confirmed to be the same. 

### How to reproduce

1. Export layers to images (PNG, rotation 90 or rotation 0 doesn't matter)
2. import layers (parameters don`t matter)

### Files

[image]
[image]
[image]
[image]
[m4_nut_m4_thumb_nut1.pws.zip](https://github.com/sn4k3/UVtools/files/14394361/m4_nut_m4_thumb_nut1.pws.zip)

sn4k3, 2024-02-24:
Will fix on next release.
But what is the complete objective? Maybe you don't need to process that outside UVtools (export/import)

yaroslaw, 2024-02-26:
I was trying to put several pieces of the same thing processed by [ScriptCompensateCrossBleeding.cs] to find out what level of compensation do I need. Firstly I tried using pattern tool and ROI to hopefully apply different level of compensation to different regions, but it didn't work as script processes everything disregarding ROI. So my next idea was to move object around, run script with diferent levels and than combine all test in one print.  

Fixing script would be more beneficial in future, but I'm just beginning to learn UVTools so don`t now much about scripting, conventions and technical limitations of script (and never touched C# in my life, but learned c++ and some other languages so probably can make some sense of a code)

sn4k3, 2024-02-29:
Fixed

closed 2024-02-29 by sn4k3

Razard, 2025-01-07:
Same issue on 5.0.3 x64 again. Trying to import the same layers to the model. Actually I need to copy the source model with move to create two models at one print. But I'm unable to import layers with any merge even if I do not move the model.

sn4k3, 2025-01-07:
@Razard This function was not changed since. Provide a sample file where I can reproduce the problem.

Razard, 2025-01-08:
[zzz.zip](https://github.com/user-attachments/files/18339782/zzz.zip)
Here I've imported 10 pictures to one file. Then tried to import the same files with merge (moving or not - does not matter).
My action: Open z025.pws, Actions -> Import Layers -> MergeMax -> Add (layer0000.png ... layer0009.png), got the same error.

sn4k3, 2025-01-08:
That is normal.
Your file is 1440 x 2560.
The images you try to import are 2560 x 1440. 
As so resolutions doesn't match

Razard, 2025-01-09:
Hmm... If I try everything from scratch.

Case A:
1. Import 10 files (not rotated) to empty project.
2. Import the same 10 files (not rotated) to already imported without modification by MergeMax option.

Case B:
0. Create rotated 10 source files.
1. Import 10 files (not rotated) to empty project.
2. Import the same 10 files (rotated) to already imported without modification by MergeMax option.

In both cases I've got the same error.

P.S. Archive attached.
layers0000-layers0009 - not rotated.
layers1000-layers1009 - rotated.
[zzz1.zip](https://github.com/user-attachments/files/18357352/zzz1.zip)

sn4k3, 2025-01-10:
It works for me with your last images (with correct resolution)

[image]

But of course, if you add every image it will fail because UVtools merge all images and creates a virtual file format with the max resolution.
In most cases you should not add images, but a file format instead. 

Also note when exporting to images you must set No Rotation and No Mirror to keep original source. UVtools auto rotates to best match the view and/or the original model, but for printer it must be raw.