sn4k3/UVtools #844: UVtools crashed due an unexpected application error.

samsadatidds, 2024-03-05:
### System

```Text
UVtools v4.2.4 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Core(TM) i5-7500T CPU @ 2.70GHz
Processor cores: 4
Memory RAM: 8.83 / 15.96 GB
Runtime: win10-x64
Framework: .NET 6.0.27
AvaloniaUI: 11.0.7
OpenCV: 4.8.1

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1040    UA: 1920 x 1040

Path:       C:\Users\Dr. Sam Sadati\Desktop\UV Tools\
Executable: C:\Users\Dr. Sam Sadati\Desktop\UV Tools\UVtools.exe
Loaded file: UVtools_demo_file.sl1s [Version: 0] [Class: SL1File]

General configuration for OpenCV 4.8.1 =====================================
  Version control:               4.8.1-255-gbb5bf05043

  Platform:
    Timestamp:                   2023-10-17T22:29:49Z
    Host:                        Windows 10.0.22000 AMD64
    CMake:                       3.23.0
    CMake generator:             Visual Studio 17 2022
    CMake build tool:            C:/Program Files/Microsoft Visual Studio/2022/Community/MSBuild/Current/Bin/amd64/MSBuild.exe
    MSVC:                        1937
    Configuration:               Debug Release MinSizeRel RelWithDebInfo

  CPU/HW features:
    Baseline:                    SSE SSE2 SSE3
      requested:                 SSE3

  C/C++:
    Built as dynamic libs?:      NO
    C++ standard:                11
    C++ Compiler:                C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe  (ver 19.37.32825.0)
    C++ flags (Release):         /DWIN32 /D_WINDOWS /W4 /GR  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /EHa /wd4127 /wd4251 /wd4324 /wd4275 /wd4512 /wd4589 /wd4819 /MP  /MD /O2 /Ob2 /DNDEBUG 
    C++ flags (Debug):           /DWIN32 /D_WINDOWS /W4 /GR  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /EHa /wd4127 /wd4251 /wd4324 /wd4275 /wd4512 /wd4589 /wd4819 /MP  /MDd /Zi /Ob0 /Od /RTC1 
    C Compiler:                  C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe
    C flags (Release):           /DWIN32 /D_WINDOWS /W3  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /MP   /MD /O2 /Ob2 /DNDEBUG 
    C flags (Debug):             /DWIN32 /D_WINDOWS /W3  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /MP /MDd /Zi /Ob0 /Od /RTC1 
    Linker flags (Release):      /machine:x64  /INCREMENTAL:NO 
    Linker flags (Debug):        /machine:x64  /debug /INCREMENTAL 
    ccache:                      NO
    Precompiled headers:         YES
    Extra dependencies:          comctl32 gdi32 ole32 setupapi ws2_32
    3rdparty dependencies:       ittnotify libjpeg-turbo libwebp libpng libtiff libopenjp2 IlmImf zlib

  OpenCV modules:
    To be built:                 core highgui imgcodecs imgproc ts videoio
    Disabled:                    dnn features2d flann gapi java ml photo python_bindings_generator python_tests video world
    Disabled by dependency:      calib3d objdetect stitching
    Unavailable:                 python2 python3
    Applications:                perf_tests
    Documentation:               NO
    Non-free algorithms:         NO

  Windows RT support:            NO

  GUI:                           WIN32UI
    Win32 UI:                    YES
    VTK support:                 NO

  Media I/O: 
    ZLib:                        build (ver 1.2.13)
[... 158 lines cut ...]
   at UVtools.UI.MainWindow.ProcessFile(String fileName, FileDecodeType fileDecodeType, UInt32 actualLayer) in UVtools.UI\MainWindow.axaml.cs:line 1673
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.SendOrPostCallbackDispatcherOperation.InvokeCore()
   at Avalonia.Threading.DispatcherOperation.Execute()
   at Avalonia.Threading.Dispatcher.ExecuteJob(DispatcherOperation job)
   at Avalonia.Threading.Dispatcher.ExecuteJobsCore(Boolean fromExplicitBackgroundProcessingCallback)
   at Avalonia.Threading.Dispatcher.Signaled()
   at Avalonia.Win32.Win32Platform.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam)
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32DispatcherImpl.RunLoop(CancellationToken cancellationToken)
   at Avalonia.Threading.DispatcherFrame.Run(IControlledDispatcherImpl impl)
   at Avalonia.Threading.Dispatcher.PushFrame(DispatcherFrame frame)
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime(AppBuilder builder, String[] args, ShutdownMode shutdownMode)
   at UVtools.UI.Program.Main(String[] args) in UVtools.UI\Program.cs:line 173
Category: Application
File: _Phrozen_test_Sonic_Mighty_4K.ctb  [Version: 4] [Class: ChituboxFile]

Machine date time: 3/5/2024 3:20:28 PM
    UTC date time: 3/5/2024 8:20:28 PM

### How to reproduce

Got to Open File, open the test file Phrozen Sonic Mini 4k test file

sn4k3, 2024-03-06:
Attach the test file

sn4k3, 2024-07-30:
No response

closed 2024-07-30 by sn4k3