sn4k3/UVtools #845: [Bug] System.IO.FileLoadException: The file checksum does not match, malformed file

farcast3r, 2024-03-06:
### System

```Text
UVtools v4.2.4 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Core(TM) i7-7700K CPU @ 4.20GHz
Processor cores: 8
Memory RAM: 13.62 / 31.93 GB
Runtime: win10-x64
Framework: .NET 6.0.27
AvaloniaUI: 11.0.7
OpenCV: 4.8.1

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (On this)
    WA: 1920 x 1040    UA: 1920 x 1040
2: 1920 x 1080 @ 100% (Primary)
    WA: 1920 x 1040    UA: 1920 x 1040

Path:       C:\Users\Jake\Downloads\UVtools_win-x64_v4.2.4\
Executable: C:\Users\Jake\Downloads\UVtools_win-x64_v4.2.4\UVtools.exe
```

### Printer and Slicer

- Printer: ELEGOO Mars Ultra 4K
- Slicer:  Chitubox Basic 2.0

### Description of the bug

I sliced a model in Chitubox but I get this error when trying to import the sliced file into UVtools:
[image]

### How to reproduce

Import sliced file (attached)
[squirtle.stl_0.05_2.5_2024_03_06_16_07_0033.ctb.zip](https://github.com/sn4k3/UVtools/files/14512597/squirtle.stl_0.05_2.5_2024_03_06_16_07_0033.ctb.zip)

Vergotear, 2024-03-07:
I have the same checksum issue for CTB files sliced with Chitubox Basic 2.0 for Saturn 2 or Mars 4 Ultra as the OP.

mostovskiy, 2024-03-08:
Same here. Elegoo Saturn 2 and Chitubox Basic 2.0

rhom6us, 2024-03-08:
Same here. Elegoo Saturn 2 and Chitubox Basic 2.0

robertorocca, 2024-03-13:
Same here too. Elegoo Mars3 and Chitubox basic 2.0

commit f3b0a047 (2024-03-14): v4.2.5

- (Add) CTB: Support for version 5, and set it as default version
- (Improvement) Encrypted CTB: Better verification of checksum hash (Fixes #845)
- (Improvement) PrusaSlicer printers that use CTB format now all strict set file version under printer notes (Please update your printers)
- (Improvement) Free unused RAM now also collects the LOH
- (Upgrade) AvaloniaUI from 11.0.7 to 11.0.10
closed 2024-03-14 by sn4k3