sn4k3/UVtools #849: [Bug]  Can't read CHITUBOX Basic V2.0 PM5S Files

jleppert, 2024-03-10:
### System

```Text
UVtools v4.2.4 X64
Operative system: Linux 5.4.0-169-generic #187-Ubuntu SMP Thu Nov 23 14:52:28 UTC 2023 X64
Processor: AMD Ryzen Threadripper 2990WX 32-Core Processor
Processor cores: 64
Memory RAM: 91.41 / 125.81 GB
Runtime: ubuntu.20.04-x64
Framework: .NET 6.0.27
AvaloniaUI: 11.0.7
OpenCV: 4.8.1

Sreens, resolution, working area, usable area:
1: 3840 x 2160 @ 150% (Primary)
    WA: 3775 x 2130    UA: 2517 x 1420
2: 2560 x 1440 @ 100%
    WA: 2560 x 1410    UA: 2560 x 1410

Path:       /tmp/.mount_UVtoolHQsAvV/usr/bin/
Executable: /tmp/.mount_UVtoolHQsAvV/usr/bin/UVtools
```

### Printer and Slicer

- Printer:  Anycubic Photon M5S
- Slicer:  CHITUBOX Basic V2.0

### Description of the bug

When opening the file, receive the following fatal error dialog box:

System.IO.FileLoadException: 'MACHINE' section expected, but got 'Photon'
   at UVtools.Core.FileFormats.PhotonWorkshopFile.AnycubicNamedTable.ValidateExpecting(Int32 length, Boolean validateName) in UVtools.Core/FileFormats/PhotonWorkshopFile.cs:line 289
   at UVtools.Core.FileFormats.PhotonWorkshopFile.DecodeInternally(OperationProgress progress) in UVtools.Core/FileFormats/PhotonWorkshopFile.cs:line 2146
   at UVtools.Core.FileFormats.FileFormat.Decode(String fileFullPath, FileDecodeType fileDecodeType, OperationProgress progress) in UVtools.Core/FileFormats/FileFormat.cs:line 4154
   at UVtools.UI.MainWindow.<>c__DisplayClass96_0.<ProcessFile>b__1() in UVtools.UI/MainWindow.axaml.cs:line 1416

### How to reproduce

Create any file with Chitubox Basic V2.0 and it crashes, file IS NOT CORRUPT

sn4k3, 2024-03-11:
This is a problem with Chitubox, they write "Photon" instead of "MACHINE" to label a section on file. While printer does not validate that it's a bug on thier side and not following the Anycubic spec. You should report this to Chitubox.

sn4k3, 2024-04-30:
Looks like fixed on new chitubox version v2.1

closed 2024-04-30 by sn4k3