sn4k3/UVtools #856: [Bug] Opening dialog Pixel Arithmetic resets layer to 1 - so can't use 'current layer' options

tlhintoq, 2024-03-25:
### System

```Text
UVtools v4.2.5 X64
Operative system: Microsoft Windows 10.0.19045 X64
Graphic card: 
Processor: 11th Gen Intel(R) Core(TM) i9-11900H @ 2.50GHz
Processor cores: 16
Memory RAM: 22.59 / 63.71 GB
Runtime: win10-x64
Framework: .NET 6.0.28
AvaloniaUI: 11.0.10
OpenCV: 4.8.1

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1040    UA: 1920 x 1040
2: 2560 x 1080 @ 100%
    WA: 2560 x 1040    UA: 2560 x 1040
3: 2560 x 1080 @ 100%
    WA: 2560 x 1040    UA: 2560 x 1040

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: 4button_yes_frame.ctb [Version: 4] [Class: CTBEncryptedFile]
```

### Printer and Slicer

- Printer:  Phrozen Mega8k
- Slicer:  Lychee 2.0 release candidate 25mar2024

### Description of the bug

Opening pixel arithmetic dialog resets the layer slider to layer 1.
Therefore options for 'current layer' in the dialog are meaningless.
Easy enough to manually enter the layer number if you noticed before opening the dialog; so this is not a critical bug that stops the program from doing its job.
Low priority.

### How to reproduce

1. Open a file - a .ctb in this example
2. Scroll through layers to find the end of the range to be affected by the next operation. 88 in this example
3.  [image]
4. Tools | Pixel Arithmetic
• Here's the bug.  The layer slider resets to layer 1
[image]

sn4k3, 2024-03-25:
That's because you have a profile set, when you set a profile it select the layer range you selected.
When you open the dialog without a profile (defaults) it does not change current layer.

tlhintoq, 2024-03-26:
Ok.  Thanks.

closed 2024-03-26 by tlhintoq