sn4k3/UVtools #857: [Bug] Can't save files since last update. (MacOS) System.NullReferenceException: Object reference not set

ok1ok1, 2024-03-28:
### System

```Text
UVtools v4.2.5 Arm64
Operative system: Darwin 23.4.0 Darwin Kernel Version 23.4.0: Fri Mar 15 00:19:22 PDT 2024; root:xnu-10063.101.17~1/RELEASE_ARM64_T8112 Arm64
Processor: Apple M2
Processor cores: 8
Memory RAM: 7.03 / 12.73 GB
Runtime: osx.14-arm64
Framework: .NET 6.0.27
AvaloniaUI: 11.0.10
OpenCV: 4.8.1

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary)
    WA: 1920 x 978    UA: 1920 x 978
2: 1920 x 1080 @ 100%
    WA: 1920 x 1055    UA: 1920 x 1055

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: test.ctb [Version: 4] [Class: ChituboxFile]
```

### Printer and Slicer

- Printer: Epax X1-4K
- Slicer: Lychee 5.4.3 & 6.0 (Beta)

### Description of the bug

The app functions normally until it comes time to save. When a save is attempted the app returns the following error.
"Error while saving the file"
System.NullReferenceException: Object reference not set to an instance of an object.
   at UVtools.Core.FileFormats.ChituboxFile.EncodeInternally(OperationProgress progress) in UVtools.Core/FileFormats/ChituboxFile.cs:line 1908
   at UVtools.Core.FileFormats.FileFormat.Encode(String fileFullPath, OperationProgress progress) in UVtools.Core/FileFormats/FileFormat.cs:line 4102
   at UVtools.Core.FileFormats.FileFormat.SaveAs(String filePath, OperationProgress progress) in UVtools.Core/FileFormats/FileFormat.cs:line 5558
   at UVtools.UI.MainWindow.<>c__DisplayClass102_0.<SaveFile>b__0() in UVtools.UI/MainWindow.axaml.cs:line 2078
   
   
[image]

### How to reproduce

1. Open a .ctb file.
2. Attempt to save the file.

I've reinstalled my slicer and UVtools. No change in behavior. This error only seems to be present on the MacOS version. 

### Files

Example file can be downloaded here:
https://memoryvault.myds.me:5001/d/s/xpfHJIUQ6ezAysmFOlJP15cGIs50eaB4/8QqtpDO3ldjwZRgRDlrStmRtuhsXK7HK-q7nAvgPaPQs

sn4k3, 2024-03-28:
The error line does not make sense to trigger the problem. Can you try downgrade 1 version and try again?

ok1ok1, 2024-03-29:
I can try that. I normally use the auto install command with terminal to upgrade. How do I install an older image?

Also, I did a little more testing. If I do a quick open and save of the .ctb file using the windows version of UV Tools 4.2.5 I can then open the file, do my edits, and save using the Mac version. Basically there is something about the files generated by Lychee that the mac version doesn't like.

sn4k3, 2024-03-29:
> I can try that. I normally use the auto install command with terminal to upgrade. How do I install an older image?

Try with: `bash -c "$(curl -fsSL https://raw.githubusercontent.com/sn4k3/UVtools/master/Scripts/install-uvtools.sh)" -- 4.2.4`

> Basically there is something about the files generated by Lychee that the mac version doesn't like.

There are no different conditions for OS-specific under UVtools core code. Which make me thing error come from UI framework.
The error line from dump does not make sense, it can not be null...

ok1ok1, 2024-03-29:
It looks like you're likely on the right track. I reverted to version 4.2.4. Everything functioned normally. 
Just to be sure I ran the 4.2.5 update again & the problem returned.

sn4k3, 2024-03-29:
As the stack trace is no useful and present the wrong cause, I have no way to debug this (No mac machine to run under debugger). Let's hope next release of the UI framework fix this issue.

pattonwebz, 2024-09-14:
I am seeing this issue (well an error with a trace that is very very similar). 

Happens when trying to convert a ctxdlpv4 file format to any of the ctb type formats.

I am running the latest release. Linux appimage version.

Stack trace:
```
System.NullReferenceException: Object reference not set to an instance of an object.
   at UVtools.Core.FileFormats.CTBEncryptedFile.EncodeInternally(OperationProgress progress) in UVtools.Core/FileFormats/CTBEncryptedFile.cs:line 1379
   at UVtools.Core.FileFormats.FileFormat.Encode(String fileFullPath, OperationProgress progress) in UVtools.Core/FileFormats/FileFormat.cs:line 4484
   at UVtools.Core.FileFormats.FileFormat.Convert(Type to, String fileFullPath, UInt32 version, OperationProgress progress) in UVtools.Core/FileFormats/FileFormat.cs:line 6404
   at UVtools.Core.FileFormats.FileFormat.Convert(FileFormat to, String fileFullPath, UInt32 version, OperationProgress progress) in UVtools.Core/FileFormats/FileFormat.cs:line 6418
   at UVtools.UI.MainWindow.<>c__DisplayClass101_0.<ConvertToOnClick>b__0() in UVtools.UI/MainWindow.axaml.cs:line 2074
```

sn4k3, 2024-09-14:
@pattonwebz Can provide the cxdlpv4 you trying to convert?

pattonwebz, 2024-09-14:
Hey @sn4k3,

It's happening with all the sliced files I have. Here's the simplest:

[cube.zip](https://github.com/user-attachments/files/17004045/cube.zip)

sn4k3, 2024-09-15:
Found the problem within the convert, it only affect version 5 of the format.
To bypass the problem go to File - Terminal and send the command: `SlicerFile.MaterialName = string.Empty;`

pattonwebz, 2024-09-15:
Amazing, thanks for looking at this so fast!

That is working after setting the empty string in terminal.

For me the issue is present when converting to both v4 and v5 ctb files. Running the command resolves the issue for both though.

https://github.com/user-attachments/assets/4d820603-9743-45bb-bdae-832bdb96431e

sn4k3, 2024-09-15:
> For me the issue is present when converting to both v4 and v5 ctb files.

After this statement I inspected the code and found that version selector have a bug causing your convertion to always use CTBv5 no matter what version you selected.

Will fix both on next release.

sn4k3, 2024-10-04:
Closing as fixed

closed 2024-10-04 by sn4k3

SirKliv, 2024-12-31:
Same error again on arm64 and intel macs ((
```
UVtools v5.0.3 Arm64
Operative system: Darwin 24.2.0 Darwin Kernel Version 24.2.0: Fri Dec  6 18:40:14 PST 2024; root:xnu-11215.61.5~2/RELEASE_ARM64_T8103 Arm64
Processor: Apple M1
Processor cores: 8
Memory RAM: 5.84 / 9.98 GB
Runtime: osx-arm64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.3
OpenCV: 4.9.0

Sreens, resolution, working area, usable area:
1: 1440 x 900 @ 100% (Primary) (On this)
    WA: 1440 x 840    UA: 1440 x 840

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: jackal_treasure_island_1.ctb [Version: 4] [Class: ChituboxFile]
```
```
UVtools v5.0.3 X64
Operative system: Darwin 24.2.0 Darwin Kernel Version 24.2.0: Fri Dec  6 18:41:43 PST 2024; root:xnu-11215.61.5~2/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz
Processor cores: 12
Memory RAM: 8.80 / 13.94 GB
Runtime: osx-x64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.3
OpenCV: 4.9.0

Sreens, resolution, working area, usable area:
1: 1792 x 1120 @ 100% (Primary) (On this)
    WA: 1792 x 1057    UA: 1792 x 1057

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: Hornet_jackal_1.ctb [Version: 4] [Class: ChituboxFile]
```
[image]
File => Convert to => Chitubox => Chitubox CTB => v1 is work, v2-v5 isn't

I tried using the terminal command above

`SlicerFile.MaterialName = string.Empty;`

but no effect.