sn4k3/UVtools #859: [Bug] M5s Pro Files saved from UVTools sometimes never stop exposing the first layer.

NightHammer1000, 2024-04-01:
### System

```Text
UVtools v4.2.5 X64
Operative system: Microsoft Windows 10.0.22631 X64
Processor: AMD Ryzen 9 7900X3D 12-Core Processor          
Processor cores: 24
Memory RAM: 24.90 / 63.62 GB
Runtime: win10-x64
Framework: .NET 6.0.28
AvaloniaUI: 11.0.10
OpenCV: 4.8.1

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 150% (Primary) (On this)
    WA: 2560 x 1380    UA: 1707 x 920

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: ContrRL.m5sp [Version: 518] [Class: PhotonWorkshopFile]
```

### Printer and Slicer

- Printer: Anycubic M5s Pro
- Slicer: Lycee Slicer 6.0 Beta, Anycubic Photon, Formware

### Description of the bug

Sometimes UVTools Produces files that Expose the First Layer for a very long time.

The Print begins normally, showing the normal Print time on the Printers Display. But as soon as the Printer starts exposing the first layer the Print Time jumps above 200 Hours and the First Layer never stops Exposing. 

File looks fine in UVTools. Also shows the right print time.
Same file not resaved in UVTools, straight out the Slicer does not have this Issue.

### How to reproduce

Does not happen all the Time. Only sometimes.

1. Slice File
2. Open File in UVTools.
3. Resave File
4. Start Print.

### Files

File edited with UVTools that experiences the Problem:
https://drive.google.com/file/d/1r8x1ZWqlfCg2-nbmCHeytiJEvz1wxz7r/view?usp=sharing

File straight out the Slicer without being touched by UVTools without the Issue:
https://drive.google.com/file/d/1a8rcMQmQv3ShW0jzvWrSrdiIjFI7jpM2/view?usp=sharing

sn4k3, 2024-04-01:
First layer is set to 30s. However some firmwares introduce problems when using `PerLayerSettings` flag, which UVtools define to true if you set any parameter different from base settings to any layer.
In your case I advice to not use UVtools to set transition layers as it can set times to layer differently than globals.
You can also reset layer settings and set them to globals by using: `File - Reset layer properties`

NightHammer1000, 2024-04-02:
So its a Firmware Issue. I hoped it would not be. 
Well off to dealing with Anycubics horrible Support again.

Gentle312, 2024-04-02:
Hey there! got the same issue with photon workshop and uv-tools. Please let me know if you find anything.

NightHammer1000, 2024-04-02:
Its definetly what @sn4k3 said.

I loaded up the Failing File. Clicked "Reset layer properties" and saved again and now its printing.

Seems like we need to hound Anycubic for a fix.

Gentle312, 2024-04-02:
> Its definetly what @sn4k3 said.
> 
> I loaded up the Failing File. Clicked "Reset layer properties" and saved again and now its printing.
> 
> Seems like we need to hound Anycubic for a fix.

Thanks for replying! Is "reset layer properties" an option in lychee or in uvtools? Couldnt find it in uvtools and usually use p-workshop or chitu for a slicer. Happy that your print did work out!

Best Regards

NightHammer1000, 2024-04-03:
I just wrote a Support Ticket to Anycubic with a Link in it to this Issue.

 @Gentle312 I urge you to do the same.

Hargrim, 2024-04-03:
Also started to happen with M5S.

sn4k3, 2024-04-03:
See past issues: https://github.com/sn4k3/UVtools/issues/742
That flag is a problem for many anycubic machines and some firmwares.

Gentle312, 2024-04-07:
> I just wrote a Support Ticket to Anycubic with a Link in it to this Issue.
> 
> @Gentle312 I urge you to do the same.

can you copy me the content of your support ticket? makes it more crystal clear.
will issue one asap

sn4k3, 2024-04-12:
Not sure what you were expecting.

If you look, which slicer does ever use or used "Per layer settings", NONE! They always follow static and fixed parameters for every bottom/normal layer. Not even on their slicer they give that option, beside file format supposedly accept it. Only UVtools takes full control of that feature, no other slicer does. So they don't really care.

They never fix, listen or care. They have bugs since first version that they never care to fix.
Thier mindset is just send cheap trash to market and make fast money. In reality we just need 2 or 4 good printer models, see how many exists on market and how fast they come out. For most people this toys are acceptable and do what they want for cheap, so user base don't care either, this is what allow bad products to survive.

This problems are easy to fix, because they are just code and can be flashed via a file, but they rather not to and send a new printer to market with the same problems.
In most printers the hardware is decent, but the firmware they code ruin all chances from being a top product. 
While other formats/printers are a little better than anycubic don't think they are any different mindset. Chitu for example ignore parameters for first layer  requiring a blank layer hack, and also per layer settings on some printers don't work to fullest.

Due the close of mSLA ecosystem, we don't have a good product all-round now. Only gcode printers can free all features for you, but they are rare and in the past they weren't the best in hardware. The only printer I ever brought was uniz ibee, due it gcode, hardware is very good too but the gcode looks buggy which I can live with.
Due all this problems I started build a open-source firmware, which I will try to put mSLA at FDM level (You can buy anyboard and a Pi) and modify as you want. The only requirement is the printer just need to have HDMI interface for it main LCD. No ETA.

NightHammer1000, 2024-04-12:
A man can dream^^

Got any recommendation for a relaible machine that does not fuss around alot and where things such as Per layer settings actually work?

sn4k3, 2024-04-12:
> Got any recommendation for a reliable machine that does not fuss around alot and where things such as Per layer settings actually work?

Very hard question, I'm not a resin entusiast, I just have 1 printer so I never tried none of the others, all input I have is from user feedback with UVtools. What I can tell you is that anycubic and creality are the worse for custom edits. And recent chitu printers had been problems with it too. Recent firmware update just disabled per layer settings on a user in CTB, so we never know.
It also depends on what you want (size and resolution), better to send a question on FB group and ask users which printer is working with the features.

[Concepts3D Athena](https://www.concepts3d.ca/) looks promissing, but I have worked in format and I don't understand yet how  can I manipulate layers. But if all features work well as promised user don't need to brother as lifts are auto.

NightHammer1000, 2024-04-12:
Thanks for the Info!

I`ll close this here as we most likely wont get a fix for it.

Thanks for your time.

closed 2024-04-12 by NightHammer1000