sn4k3/UVtools #870: [Bug]  Can't use Reset Layer Properties to get PerLayerOverride to False on file create with ChituBox 2.1

JoveToo, 2024-04-29:
### System

```Text
UVtools v4.3.2 X64
Operative system: Linux 6.8.7-300.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Apr 17 19:21:08 UTC 2024 X64
Processor: AMD Ryzen 9 7950X3D 16-Core Processor
Processor cores: 32
Memory RAM: 12.77 / 30.55 GB
Runtime: fedora.40-x64
Framework: .NET 6.0.29
AvaloniaUI: 11.0.10
OpenCV: 4.8.1

Sreens, resolution, working area, usable area:
1: 3840 x 2160 @ 150% (Primary)
    WA: 3840 x 2113    UA: 2560 x 1409
2: 3840 x 2160 @ 150%
    WA: 3840 x 2113    UA: 2560 x 1409
3: 3840 x 2160 @ 150%
    WA: 3840 x 2113    UA: 2560 x 1409

Path:       /tmp/.mount_UVToolbJhIJv/usr/bin/
Executable: /tmp/.mount_UVToolbJhIJv/usr/bin/UVtools
Loaded file: UVtools_demo_file.sl1s [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer:  AnyCubic M5S
- Slicer:  Chitubox 2.1.0

### Description of the bug

the M5S has a bug that it fails to print files with the PerLayerOverride settings. Previous advise was to he "Reset Layer Properties" before a save, to force PerLayerOverride to false. 
This is not working anymore.

### How to reproduce

Generate a file with ChituBox 2.1.0
Work on it in UVTools

sn4k3, 2024-04-29:
Provide a test file with that behavior. I'm not seeing how slicer can glitch the reset.
Also check if you have auto applied suggestions, I think the transition layer suggestion can trigger a true on that flag.

JoveToo, 2024-04-30:
How do I check the auto applied suggestions?

I have several times closed and reopened the file to try the "Reset Layer Properties"

sn4k3, 2024-04-30:
They are set by you, under suggestion settings see if the check is enabled:

[image]

JoveToo, 2024-04-30:
[image]

sn4k3, 2024-04-30:
Check if that is also disabled on all other wait time suggestions

JoveToo, 2024-04-30:
I do see this:

[image]

sn4k3, 2024-04-30:
That would not affect. I will have a look into your file.

sn4k3, 2024-04-30:
Problem is related to chitubox, they have one layer (1035) with different heigth, possible due rounding errors:

[image]

They should all be 0.05, but that 0.049 is forcing flag to true, to respect the height on that layer as it is different from global.
That's the reason the reset wont work, it don't mess with position, only with print settings.

To force all layers to be re-built (settings and height) you go to File - Terminal, and send the command: `SlicerFile.RebuildLayersProperties()` that will recalculate everything from globals.

closed 2024-07-30 by sn4k3

epareja2, 2024-09-01:
Hello,
I've read multiple issues on Anycubic printers not working with the layer exposure times, and the solutions seems to Reset the layer properties.

I have an Anycubic M5s and I'm using Chitubox 2.2 as the slicing software.
I tried to reset the layers but I still get the "PerLayerOverride" = "True".

I also tried to run the terminal with the command SlicerFile.RebuildLayersProperties() but nothing happens.

What is the procedure I should check if the file is generating correctly?

My problem is that the base layers work but the normal layers get a 0s exposure time.

Thank you

sn4k3, 2024-09-01:
Check if you have any suggestion with auto-apply enabled. If so it will always change the properties.

epareja2, 2024-09-02:
Thank you for your answer.
It seems all are turned off
[image]
[image]
[image]

I tried to upload a new file from Chitubox and the flag is always false. I have processed all the succestions (islands, traps and suction sections, and the flag remains false.

I will try this with a very small file to test i it works.

Is there a way to open the result file and check something before starting the print and find out that it didn't work?

Thank you

sn4k3, 2024-09-02:
About the other file, it can be due slicer rounding problems, like this: https://github.com/sn4k3/UVtools/issues/870#issuecomment-2086403545

> Is there a way to open the result file and check something before starting the print and find out that it didn't work?

What did you mean?

epareja2, 2024-09-02:
I mean that I can check the exposure time for normal layers is not going to be 0 as it has happened to me the two times I've tried to print a UVTools repaired file. Or checking the flag PerLayerOverride as False, it's the only confirmation one needs to be confident that the exposure time won't be 0?

sn4k3, 2024-09-02:
`PerLayerOverride` flag can be checked just after you save the file with UVtools, the save action is the only thing will change the flag.

sn4k3, 2024-09-02:
In any case you can always force full reset using the reset layers button while holding SHIFT, that will rebuild everything, including hight and solve round problems

epareja2, 2024-09-02:
I did that several times and also run the command in the terminal, but the flag always remained to TRUE. This didn't happen when I started processing the same fresh file though.

sn4k3, 2024-09-02:
Did you run it as last operation before save?

epareja2, 2024-09-02:
I think I did after saving, because I started printing and it failed in the normal layers so I decided to search for this problem in google, so I think I didn't do any further modifications. 
I tried to Reset Layer Properties with Shift and the terminal command, but the flag didn't change to False. 
After your answer, I did load the original file and tried to fix automatically all the suggestions, and the flag was always False.

JoveToo, 2024-09-02:
> I tried to Reset Layer Properties with Shift and the terminal command, but the flag didn't change to False.

The flag only changes to False if you save.

sn4k3, 2024-09-02:
> > I tried to Reset Layer Properties with Shift and the terminal command, but the flag didn't change to False.
> 
> The flag only changes to False if you save.

This is correct.
The only way to check for flag current status without save is using Terminal with a command

epareja2, 2024-09-02:
I've run a test saving the file and the flag was false before saving, and after saving the flag is true.

[image]

I reseted the layers properties, but the flag still remains in TRUE.
[image]

sn4k3, 2024-09-02:
In that case you can send the file for I review, send the original only.

epareja2, 2024-09-02:
Thank you... You can download the original exported .pm5s file.
https://wetransfer.com/downloads/356f88c9069ce840c372a62f64aec66a20240902195953/697989ad39f7063ccb641bf26c48998920240902195953/5aa7b6?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid

sn4k3, 2024-09-02:
Your file works to me, where I reset properties and after save flag is false, see:
[image]

epareja2, 2024-09-02:
Yes, you are right. I reopened the file again and it has the flag = true on that moment, and after running the command, it's getting to False. I don't know what I did yesterday as I executed the same procedure. The only command I didn't run was SlicerFile.UsingPerLayerSettings to display the flag value, but I can see it's synchronized with the left panels value.

Thank you very much @sn4k3. I will try to print this file.