sn4k3/UVtools #871: [FEATURE REQUEST]  Issue fixing: keyboard shortcuts

JoveToo, 2024-04-30:
### Is your feature request related to a problem?

Currently, navigating through a file requries the use of the mouse. It means i have to constantely move the mouse all over the screen for all the relevant buttons when fixing defects.

### Describe the solution you'd like

I would like to see keyboard shortcuts for the common things you do when fixing defects.
I would propose:

For moving through the current selected layer:
A (move Left -- scroll layer right)
D (imove Right -- scroll layer left)
W (move up -- scroll layer down)
S (move down -- scroll layer up)
These are common key combinations used in many games for the same functionality. Should feel natural.

For  moving through the layers:
Q (layer up)
E (layer down)

and lastly:
R (Apply pixel editor changes)
F (Undo last pixel editor action -- this is new functionality and doesn't exist yet)

This would allow me to do all the relevant actions during fixing defects with a single hand while keeping the mouse mostly to the area where I am working.

sn4k3, 2024-04-30:
I will look into this.
To apply pixel editor changes: CTRL + E

sn4k3, 2024-07-30:
Implemented

closed 2024-07-30 by sn4k3