sn4k3/UVtools #884: [Bug] Build plate doesn't lower to zero to print first layer Phrozen Sonic Mini 8K S (PRZ)

CrusherJoe, 2024-06-10:
### System

```Text
UVtools v4.3.2 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz
Processor cores: 8
Memory RAM: 8.98 / 15.96 GB
Runtime: win10-x64
Framework: .NET 6.0.29
AvaloniaUI: 11.0.10
OpenCV: 4.8.1

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1050    UA: 1920 x 1050
2: 1920 x 1080 @ 100%
    WA: 1920 x 1050    UA: 1920 x 1050

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: UVtools_demo_file.sl1s [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer:  Phrozen Sonic Mini 8ks
- Slicer: Lychee 6.0.2

### Description of the bug

Import .prz into UVtools and remove islands. Accept suggestions for wait times, number of bottom layers, etc. Save file. Attempt to print on printer. When print starts, the plate lowers into the vat, instead of exposing the first layer, it very slowly rises along the z-axis, after which it lowers even more slowly back into the vat, but not near enough to zero, then starts printing even though the plate is at a considerable distance from the screen, and everything remains on the film. It either does this or some variation, the end result it the plate never gets to zero to expose the first layer. Yes, it's doing a "blank" first layer but doesn't lower the plate to do the first "non-blank" first layer.

Here is a video of the behavior: https://photos.app.goo.gl/1phGKzbAtNL2L3xg7

### How to reproduce

Import .prz to UVtools, prepare file, export, attempt to print.

sn4k3, 2024-06-10:
Try to remove VAT and place a papper. Print same file again and original without UVtools to confirm.

CrusherJoe, 2024-06-10:
Ah, geez, why didn't I think of that. With a freshly sliced file the behavior continues. The issue is not with UVtools! I'll close this issue for you!

Thank you for your help!

closed 2024-06-10 by CrusherJoe

sn4k3, 2024-06-10:
Yeah, when something is odd, always try original file. These machines firmwares are very bad, by now I expect everything...