sn4k3/UVtools #891: AppImage 4.3.2 appears to be missing a directory

Tinker012, 2024-06-23:
### System

```Text
Operating System is Ubuntu 42.04 LTS

I am unable to sucessfully run the AppImage file which 
is identified as UVtools_linux-x64_v4.3.2.AppImage
```

### Printer and Slicer

- Printer: N/A
- Slicer: N/A

### Description of the bug

I am attempting to run the latest AppImage 4.3.2 on an installation of Ubuntu 24.04 destop LTS with AppImageLauncher installed (I am able to sucessfully run other Appimage files on this system) and I am receiving a system notification- 
                Failed to launch "UVTools"
                Failed to change to directory "~" (No such file of directory)

### How to reproduce

click on Show Apps
click on icon UVtools

sn4k3, 2024-06-23:
I have notice that on recent systems.
`~` definitely exists as it your home directory however somehow it fails.
I also note that the problem is with AppImage integration on your system, if you run it from desktop without integrate it always run fine.

I don't have a clue about the problem, but I will try to find anwsers.

Tinker012, 2024-06-23:
Thank you, I hadn't thought of trying to run the AppImage without the AppImageLauncher integration.

I very much appreciate your response

sn4k3, 2024-07-30:
Fixed

closed 2024-07-30 by sn4k3