sn4k3/UVtools #896: [FEATURE REQUEST] consider transition layers as bottom layers for purposes of SB/LB/SN/LN navigation

rinqu-eu, 2024-07-04:
### Is your feature request related to a problem?

When using SB/LB/SN/LN navigation;
if the raft is tall enough to be split across the bottom layers and the transition layers
using either LB or LN will most likely return one of the raft layers.

Example: when we have 5 bottom layers and 10 transition layers. Both return the raft as being the biggest mass
[image]

### Describe the solution you'd like

1st choice: Marking transition layers as 'bottom' instead of 'normal'.
2nd choice: add a separate category for transition layers, and add ST and LT navigation.

Example: when we change the bottom layers count to 15 to simulate transition layers being included, LN actually returns part of the print as the biggest mass.
[image]

sn4k3, 2024-07-05:
If the raft is part of both bottom and normal layers and is the largest mass on the file that is the expected behavior and returning exactly what it should, this not a issue and the behavior will not change regarding the terms SB/LB/SN/LN. It foreach the layer group the return the most pixel layer.

There is no absolute way to detect what is a raft/support/model which also vary from user to user / print to print.
If you want to know the largest mass you can code a snippet code to run on `File - Terminal` which ignore first n layers defined by you. So if your rafts are never taller than x you can skip that layers and compute with the rest.

rinqu-eu, 2024-07-05:
I understand that this is an intended behaviour and it is returning exactly what it's supposed to.
I didn't suggest autodetecting rafts/supports/models. I've only suggested that transition layers get treated as bottom (as in IsBottomLayer = true) or separate 'transition' layers.

Looking at the code now, FirstTransitionLayer and FirstNormalLayer seem to return the same thing, where in my mind, I'd assume that the order goes bottom -> transition -> normal

I'm basically suggesting, that transition layers should be a separate category, instead of having overlap with normal layers based on the fact that they have a different exposure time.

sn4k3, 2024-07-05:
In common case the transition layers starts on first normal layer and goes on fading. It does not create a gap between bottom and normals "group".
Bottom and normal layers are just terms to differentiate two groups of settings. 
As for code is safer to just have two groups that always exists, than 3 groups that 1 could not, that will remain un-change.

In terminal you can calculate what you suggest by add the first normal with transition layer count

rinqu-eu, 2024-07-06:
I will implement my case via the terminal then, thank you

closed 2024-07-06 by rinqu-eu