sn4k3/UVtools #897: AppImage doesnt run outside its directory

lordekeen, 2024-07-05:
### System

```Text
Distro: Rocky Linux 9.4 (RHEL 9)
DE: GNOME 40
```

### Printer and Slicer

- Printer: Elegoo Mars 2 Pro
- Slicer: Chitubox

### Description of the bug

When running the app in its directory, it opens normally, but from a .desktop launcher (so it shows in the menu) it does not. I did integrated it into the system using Gear Lever (a Flatpak on Flathub) which does a simple desktop integration (extract the icon, generate the .desktop and put it into ~/.local/share/applications). 

### How to reproduce

1. Download the AppImage
2. Make it executable as a program
3. Create a .desktop file in ~/.local/share/applications
4. Edit the .desktop pointing to the AppImage (exec=/path-to-appimage)
5. Try launching from the menu

sn4k3, 2024-07-05:
I followed your workflow, but it works to me:
[image]

lordekeen, 2024-07-06:
Maybe I've messed something up, thanks for looking into it, closing the issue.

closed 2024-07-06 by lordekeen