sn4k3/UVtools #899: [FEATURE REQUEST] 8 bit grey scale image for dl2p format

Veenxz, 2024-07-08:
### Is your feature request related to a problem?

Hi,

I am working on 4D printing for hydrogels, now I can only get 16 grey values using UV Tool. Would it be possbile for me define more light intensity on my pattern?

Thanks for the wounderful tool!

### Describe the solution you'd like

+ Add LayerImageColorTable for 8 bit

### Files

[UVtools.Core/FileFormats/PhotonWorkshopFile.cs](https://github.com/sn4k3/UVtools/blob/2cf889e8b8743f8263ef7dc46a370d914df86822/UVtools.Core/FileFormats/PhotonWorkshopFile.cs#L508)
```C#
    public sealed class LayerImageColorTable
    {
        [FieldOrder(0)] public uint UseFullGreyscale { get; set; }
        [FieldOrder(1)] public uint GreyMaxCount { get; set; } = 16;
        [FieldOrder(2)] [FieldCount(nameof(GreyMaxCount))] public byte[] Grey { get; set; } = {
            // AA16: 255, 239, 223, 207, 191, 175, 159, 143, 127, 111, 95, 79, 63, 47, 31, 15
            15, 31, 47,    // 1,2,3
            63, 79, 95,    // 4,5,6
            111, 127, 143, // 7,8,9
            159, 175, 191, // 10,11,12
            207, 223, 239, // 13,14,15
            byte.MaxValue  // 16
        };

        [FieldOrder(3)] public uint Unknown { get; set; }

        public override string ToString()
        {
            return $"{nameof(UseFullGreyscale)}: {UseFullGreyscale}, {nameof(GreyMaxCount)}: {GreyMaxCount}, {nameof(Grey)}: {Grey}, {nameof(Unknown)}: {Unknown}";
        }
    }
```

Veenxz, 2024-07-09:
I'm using the Anycubic D2 DLP printer, I have tested the machine and could adjust the exposure intensity according to the grey scale. The max brightness in UVTools is 255, I didn't find how to set a higher AA level in the software.
BTW, could I adjust the exposure time and greyscale in one slice?

If I import this file in UVTools, the saved file will only get 16 greyscales. [Intensity_test.zip](https://github.com/user-attachments/files/16135980/Intensity.zip)

The dl2p file: [0-100-delta2_light_1.zip](https://github.com/user-attachments/files/16136093/0-100-delta2_light_1.zip)

You can also see, in the test, that some patterns have converged to the same greyscale.

[image]
[image]

sn4k3, 2024-07-09:
> The dl2p file: [0-100-delta2_light_1.zip](https://github.com/user-attachments/files/16136093/0-100-delta2_light_1.zip)

You can also see, in the test, that some patterns have converged to the same greyscale.

That file uses AA=1 defined on file, if it does gray values firmware may ignore all information on it and convert grays directly.

> The max brightness in UVTools is 255

Thats the max pixel color of every lcd, color or mono max is 255 per pixel. You can't use higher values.

Veenxz, 2024-07-09:
Got it. Thanks for your reply!

closed 2024-07-09 by Veenxz