sn4k3/UVtools #903: thumbnail not showing after saving PHZ with UVtools

bordermultimedia, 2024-07-19:
### System

```Text
UVtools v4.3.2 X64
Operative system: Microsoft Windows 10.0.22631 X64
Processor: 11th Gen Intel(R) Core(TM) i7-11700K @ 3.60GHz
Processor cores: 16
Memory RAM: 42.78 / 127.81 GB
Runtime: win10-x64
Framework: .NET 6.0.29
AvaloniaUI: 11.0.10
OpenCV: 4.8.1

Sreens, resolution, working area, usable area:
1: 3840 x 2160 @ 175% (Primary) (On this)
    WA: 3840 x 2076    UA: 2194 x 1186
2: 1680 x 1050 @ 100%
    WA: 1680 x 1002    UA: 1680 x 1002

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: tanque-Chasis_06h46m_44ml_copy.phz [Version: 2] [Class: PHZFile]
```

### Printer and Slicer

- Printer: Prozen Sonic Mini
- Slicer: Lychee

### Description of the bug

Lychee produces a PHZ file that has a thumbnail, I can see it on the printer's screen, but whenever I open it with UVtools, and then save it again (even if I did not make any fixes or modifications) it saves a PHZ file that doesn't show any thumbnail on the printer. Even opening it again with UVtools shows a corrupt image. Also, the new file is bigger, but I suppose that has something to do with compression settings.

### How to reproduce

open any PHZ file with a visible thumbnail and save as a new file, open the new file to see if the thumbnail is still present. I'm sending you sample files to show this issue.

### Files

[cube.zip](https://github.com/user-attachments/files/16314499/cube.zip)
[cube_48m_1ml_copy.zip](https://github.com/user-attachments/files/16314500/cube_48m_1ml_copy.zip)

sn4k3, 2024-07-19:
Will check and try to solve in next patch.

sn4k3, 2024-07-30:
Fixed

closed 2024-07-30 by sn4k3