sn4k3/UVtools #905: [Bug]  Pixel arithmetic error on corrode

tlhintoq, 2024-07-24:
### System

```Text
UVtools v4.3.2 X64
Operative system: Microsoft Windows 10.0.19045 X64
Graphic card: 
Processor: 11th Gen Intel(R) Core(TM) i9-11900H @ 2.50GHz
Processor cores: 16
Memory RAM: 32.38 / 63.71 GB
Runtime: win10-x64
Framework: .NET 6.0.29
AvaloniaUI: 11.0.10
OpenCV: 4.8.1

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: 24jul.ctb [Version: 4] [Class: CTBEncryptedFile]
```

### Printer and Slicer

- Printer:  Elegoo Saturn-S
- Slicer: Lychee 6.0.2

### Description of the bug

[image]

### How to reproduce

File sliced from Lychee
Select `Pixel Arithmatic` - all layers, then change first layer to 1 instead of 0
Go
[image]

### Files

https://drive.google.com/drive/folders/1OTiMN7uHQ7uwOQytJx8_mgt_bE9vCK0b?usp=sharing

sn4k3, 2024-07-24:
I copied exactly your settings but does not raise for me.
Can you export your operation settings and attach here the uvtop file?

tlhintoq, 2024-07-24:
@sn4k3 
You lost me.
"operation settings" and "uvtop file"?
Let me know what you want and were I find it, and I'll be happy to include it

sn4k3, 2024-07-24:
On tool dialog, bottom left, click the button and export the settings. Attach here the file as txt.

tlhintoq, 2024-07-25:
I'll work on that.  But here's a way I've been able to reproduce it consistently.

Perform a corrode - it succeeds.
Undo
Perform the corrode again, exact same settings - ERROR
Perform the corrode again - it succeeds.

It's almost like something isn't getting reset causing it to fail every-other time

tlhintoq, 2024-07-25:
[PixelArithmetic.zip](https://github.com/user-attachments/files/16368636/PixelArithmetic.zip)

tlhintoq, 2024-07-25:
Another observation that will cause it to happen:
Perform a corrode
Then undo
Then try to do it again
It seems to fail every-other-time.
Kinda feels like something isn't getting cleared, purged.

sn4k3, 2024-07-25:
None of that attempts trigger the problem to me.
The file you provided have no empty layer, but I created one on top, still passes... 

I know there is a bug on Undo/Redo that does not revert the exact change but still I can not reproduce it on your file.

tlhintoq, 2024-07-25:
Hmmm... Well... If you can't reproduce it then you can't.  Its pretty common for me.
Is there something else I can get you?  Registry files or config for the application?

sn4k3, 2024-07-25:
The trace is not useful as it show the wrong line.
The best bet to catch the error is by attach visual studio to debug it.

When problem happens you can also screenshot the Log tab in UVtools, to see what operations were ran. I tried every your combination and still can't reproduce, in either stable or my current dev both is working. In fact I fixed this symptom before with ROI some releases ago, I have no clue where is it leaking because this error happen with a empty ROI or a out of bounds ROI. Since function return full layer size when a empty roi is detected i'm clueless in this one.

sn4k3, 2024-07-30:
Try new version and report back

tlhintoq, 2024-08-03:
> Try new version and report back

Half a dozen jobs yesterday.  Similar in size.  Exact same product (different personaliztion).  And totally different items that get the same treatment.  No crashes.
I don't know what you found, but I think it can be called 'fixed'.  Worst case is if it happens again, then a new ticket.

Thank you SO MUCH for handling that!

closed 2024-08-03 by tlhintoq

sn4k3, 2024-08-03:
I did nothing other than put a check to not go further when find a empty layer but that check was also present later on.
Other than that libraries have been updated. Still I'm not confident that the problem is truly gone.