sn4k3/UVtools #906: Saturn 4 Ultra needs all speeds set to 0.05 mm/min for the tilt vat.[Bug]

krosnyak, 2024-07-24:
### System

```Text
UVtools v4.3.2 X64
Operative system: Microsoft Windows 10.0.22631 X64
Processor: AMD Ryzen 5 5600 6-Core Processor              
Processor cores: 12
Memory RAM: 6.46 / 31.92 GB
Runtime: win10-x64
Framework: .NET 6.0.29
AvaloniaUI: 11.0.10
OpenCV: 4.8.1

Sreens, resolution, working area, usable area:
1: 3840 x 2160 @ 150% (Primary) (On this)
    WA: 3840 x 2088    UA: 2560 x 1392

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: UVtools_demo_file.sl1s [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: Elegoo Saturn 4 Ultra
- Slicer: Chitubox Basic, Formware 3d.

### Description of the bug

The problem is that UVTools Edit Print Parameters Tool will not let me enter speed values below 10mm per minute. Elegoo Saturn 4 Ultra with the new tilt vat hardware, needs all speeds set to 0.05 mm/min since it offers only 2 preset speeds selectable on the printer UI. When other speeds are on the slice file, the machine prints very slowly.

[image]

### How to reproduce

go to edit parameters tool in UV Tools, try to enter speed values below 10mm/min. It will bump the value to 10mm/min.

sn4k3, 2024-07-24:
Also being discussed here: https://www.facebook.com/groups/uvtools/posts/1135010564275112/

> needs all speeds set to 0.05 mm/min

I find that the slicer sets the speeds and lift to whatever layer height you slice, so if you slice with 0.1mm layers it will set all to 0.1mm.

>  it offers only 2 preset speeds selectable on the printer UI

Can you post the that options here?

On next release I will decrease the min limit to account this problem.

sn4k3, 2024-07-30:
Fixed

closed 2024-07-30 by sn4k3

krosnyak, 2024-07-31:
Thank you so much!
Gonzalo