sn4k3/UVtools #909: [Bug] saving file through "Save as..." with no extension by default, making it impossible to save it again.

PashaWNN, 2024-07-27:
### System

```Text
UVtools v4.0.2 X64
Operative system: Darwin 23.6.0 Darwin Kernel Version 23.6.0: Thu Jun 20 00:24:53 PDT 2024; root:xnu-10063.140.29.0.1~15/RELEASE_ARM64_T8103 X64
Processor: Apple M1
Processor cores: 8
Memory RAM: 5.07 / 8.23 GB
Runtime: osx.14-x64
Framework: .NET 6.0.22
AvaloniaUI: 11.0.4
OpenCV: 4.8.0

Sreens, resolution, working area, usable area:
1: 1440 x 900 @ 100% (Primary)
    WA: 1440 x 813    UA: 1440 x 813

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: test [Version: 516] [Class: PhotonWorkshopFile]
```

### Printer and Slicer

- Printer: Anycubic Photon Mono 4K
- Slicer: LycheeSlicer

### Description of the bug

"Save as..." dialog has no file extension specified by default. Saving the file without extension by default makes saving it again impossible.

Expected behavior after saving file without extension:
Save dialog opens, allowing to specify filename

Actual behavior:
"Unable to find the target extension." message is shown.

### How to reproduce

1. Open file with ".pwma" extension
2. Edit it
3. Save it through "Save as..." without extension (possibly by mistake as it has no extension in save dialog by default)
4. Try to save it again using "Save as..."

sn4k3, 2024-07-27:
In fact I do no validation to file name and solely trust on system dialog scheme.

**Windows:** Filename show with extension on dialog, removing will force the extension, changing will make it append the real extension in the end, so in this case it's impossible to mess it up.
**Linux:** Filename shows with no extension on dialog, but saving will append it, but if you put a new extension in filename it will use that instead make it wrong.
**macOS:** Works like windows but extension is not displayed on filename, this also proved to be impossible to temper in my tests (Monterey)

So in this case I'm observing a different behavior than you. Probably due different system I don't know.
Here my two tests that yield in correct extension output:

[image]
[image]

PashaWNN, 2024-07-29:
Yeah, that is fixed in new version. After upgrade I've got "file format" dropdown in save dialog along with automatically appended extension by default and it is appending extension during save even if I manually delete it. Thanks.

closed 2024-07-29 by PashaWNN