sn4k3/UVtools #910: [Bug] Crashes MacOS - computer freezes and restarts

ServerOnTheMove, 2024-08-02:
### System

```Text
UVtools v4.4.0 X64
Operative system: Darwin 23.6.0 Darwin Kernel Version 23.6.0: Fri Jul  5 17:53:24 PDT 2024; root:xnu-10063.141.1~2/RELEASE_ARM64_T6020 X64
Processor: Apple M2 Pro
Processor cores: 12
Memory RAM: 10.24 / 31.48 GB
Runtime: osx.14-x64
Framework: .NET 6.0.32
AvaloniaUI: 11.1.1
OpenCV: 4.9.0

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: UVtools_demo_file.sl1s [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: Anycubic Mono M5s
- Slicer: LycheeSlicer

### Description of the bug

Loading the file works fine; the First detection takes over an hour, and the second detection (after ignoring some issues) results in the Computer freezing and then crashing/restarting. 

I was able to reproduce the issue with only UVTool running. Never had MacOS crash/restart itself.

### How to reproduce

Load file; Detect Issues; Mark Suction Cups as ignore; Detect Again -> Freeze, Crash, Restart

### Files

https://1drv.ms/u/s!Ap87vvDWL0D_kIxP3RnRaEb3w5hGCg?e=M4wkoC

ServerOnTheMove, 2024-08-02:
Diag file - I think it is from crash(es)
[Archive.zip](https://github.com/user-attachments/files/16465428/Archive.zip)

ServerOnTheMove, 2024-08-02:
I saved the file after the first detect round. When I load that saved file, the issue happens very soon after starting detection.
I am linking the file here - this is the saved file after first detection, file shared in bug report is before the first detection.

https://1drv.ms/u/s!Ap87vvDWL0D_kIxQpEbRDCj-wqRkEA?e=10e5yZ

ServerOnTheMove, 2024-08-02:
Just confirmed - 4.3.2 does not have any issues and completes successful.

sn4k3, 2024-08-02:
I'm working the file as you suggest.
In meanwhile the file, all layers are equal the raft, is this the same file you tested for?

ServerOnTheMove, 2024-08-02:
I just noticed that all layers are equal. It happens sometimes, but I believe this having the first layer replicated all they way through an issue with Lychee. I haven't had it in a while.

sn4k3, 2024-08-02:
Anyway I tried but it success without problem.
[image]

v4.4 update didn't touch issue detection.
You can try to remove UVtools from apps and reinstall using the auto installer to make sure all files are from newest version.

ServerOnTheMove, 2024-08-02:
Hi, I did try the reinstall already as I was trying to get to the root cause. After my 4.3.2 test, I went back with a full reinstall.
The interesting part is that it only happens after the second round of "Detection". First round works fine, Second one is where the freeze/restart happens. Interesting for me was that if I save after the first detection, restart the computer (just as a measure), the nload the saved file and hit detection - it happens again.

Thanks for trying to reproduce - I keep an eye on it. Not much we can do, if it is an edge case.

sn4k3, 2024-08-02:
Can you try to monitor RAM while it runs? The only cause I think off is lack of RAM, no other thing can mess like that with system.

GingerNinjaUk, 2024-09-03:
I am also getting similar issues, I upgraded to 4.4.1 a couple of days ago and ever since I did UVTools has not worked and keeps killng my Mac, it starts to do the first run (sometimes completes, but always fails when it trys to find issues), the Memory just keeps growing until I get this....

[image]

sn4k3, 2024-09-03:
> I am also getting similar issues, I upgraded to 4.4.1 a couple of days ago and ever since I did UVTools has not worked and keeps killng my Mac, it starts to do the first run (sometimes completes, but always fails when it trys to find issues), the Memory just keeps growing until I get this....

Your problem is related to a memory leak, copy and past your system details from Help - About and provide the exact file cause this trouble.

closed 2024-12-03 by sn4k3