sn4k3/UVtools #911: Image Gets Horizontally Flipped When Converting Chitubox File to CXDLP V2 Format [Bug]

ottoglass, 2024-08-03:
### System

```Text
UVtools v4.4.0 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Core(TM) i7-4790K CPU @ 4.00GHz
Processor cores: 8
Memory RAM: 8.52 / 15.88 GB
Runtime: win10-x64
Framework: .NET 6.0.32
AvaloniaUI: 11.1.1
OpenCV: 4.9.0

Sreens, resolution, working area, usable area:
1: 1080 x 1920 @ 100%
    WA: 1080 x 1880    UA: 1080 x 1880
2: 3840 x 2160 @ 150% (Primary) (On this)
    WA: 3840 x 2100    UA: 2560 x 1400

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: UVtools_demo_file.sl1s [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: Halot One (latest firmware)
- Slicer: Chitubox v1.9.5

### Description of the bug

When converting a Chitubox file to the CXDLP V2 format, the image gets horizontally flipped, resulting in a mirrored print.

### How to reproduce

Steps to Reproduce:

Open a Chitubox file.
Convert the file to CXDLP V2 format.
Inspect the output file or proceed to print.
Notice that the image is horizontally flipped in the CXDLP V2 file.

Expected Behavior:
The image should retain its original orientation after converting from Chitubox to CXDLP V2, without any horizontal flipping.

Actual Behavior:
The image is horizontally flipped, causing the printed result to be mirrored compared to the original design.

Additional Context:
This issue leads to incorrect prints, which is particularly problematic for designs where orientation is crucial.

Attachments:

-Screenshot of the original Chitubox file
[image]

-Screenshot of the CXDLP V2 file showing the flipped image
[image]
-Photos of the printed result showing the mirrored print
[image]

-ctb file
Please look into this issue and provide a fix or workaround as soon as possible. Thank you.

### Files

[_EVA3_H2_Bottom_Volcano.zip](https://github.com/user-attachments/files/16482356/_EVA3_H2_Bottom_Volcano.zip)

sn4k3, 2024-08-03:
UVtools never modify your layer image without your command. Converting does convert one to other format keeping all images.
The flip is set by slicer, different printers require different flip (None or vertical or horizontal).
If you use a CTB file to convert to other format you must slice that CTB in the correct flip format for your output printer.

Also UVtools virtual flip your layer image to show original image without flip OR to best fit into window, as that flip is virtual it is never  applied or affect the final image. You can toogle the Flip and rotate button to be grey and you will see same image as the printer will.

Here your problem:
[image]

You sliced your CTB as horizontal mirror layers but your CXDLP printer requires no mirror, hence creating your problem. You printing mirror layers into a printer that must print without mirror.
To fix that go to your slicer and select Mirror/Projector type to None.

ottoglass, 2024-08-03:
Thank you for your detailed explanation. You are correct, the issue was in the slicer settings. Thanks for clarifying how to avoid this problem going forward

closed 2024-08-03 by ottoglass

sn4k3, 2024-08-03:
Flag is just informative, in case of [CTB](https://github.com/sn4k3/UVtools/blob/3ad3524f1fb7785c7f063dbf88d6bb93eafe8762/UVtools.Core/FileFormats/CTBEncryptedFile.cs#L752-L760) it's recorded on file, while other formats does not have that information like CXDLP because looks like all their printers use same "mirror" logic. However none of this matter to printers, they will always show the image you saved to file, if the image is in wrong orientation printer doesn't care, this is all at the care of user&slicer which must set the correct orientation in first place. This also means the slicer can simply save mirror images but record the mirror flag as None (Software bug) and still it will print fine.

Is possible to know which mirror a printer uses by open the original printer on chitubox and look at LCD Mirror value. If not exists you can look at UVtools printers database.

> Is it possible to generate a warning when converting to a format that does not support mirroring?

With the current UVtools implementation, no. It would require some code change and that would only be possible on file formats that only have 1 true mandatory orientation. In your case CTB -> CXDLP v2 fits as CTB can have multiple orientations but CXDLP always expect no flip. But if you look the otherwise CXDLP > CTB is not possible to predict if the orientation is correct, because it supports all orientations so all are valid for the file even if you have the wrong one for the printer. 
Also note a file format that lacks mirror flag doesn't mean is not suppose to mirror, for example, anycubic file format does not have the flag but all layers need horizontal mirror for their machines.
Consulting UVtools printer database could be an option but is a waste of time because custom printers on Chitubox are output as "default" as machine name, which make it impossible to know the target printer by name.