sn4k3/UVtools #912: [Bug] Error when opening file - Emgu.CV.Util.CvException: OpenCV: !_src.empty()

VasilNikolov, 2024-08-11:
### System

```Text
UVtools v4.4.0 X64
Operative system: Microsoft Windows 10.0.22621 X64
Processor: AMD Ryzen 5 1600 Six-Core Processor            
Processor cores: 12
Memory RAM: 7.78 / 15.95 GB
Runtime: win10-x64
Framework: .NET 6.0.32
AvaloniaUI: 11.1.1
OpenCV: 4.9.0

Sreens, resolution, working area, usable area:
1: 3840 x 2160 @ 150% (Primary) (On this)
    WA: 3840 x 2088    UA: 2560 x 1392

Path:       C:\Users\vasil\Downloads\UVtools_win-x64_v4.4.0\
Executable: C:\Users\vasil\Downloads\UVtools_win-x64_v4.4.0\UVtools.exe
```

### Printer and Slicer

- Printer: 
- Slicer: Prusaslicer 2.7.3

### Description of the bug

The following error message appears when opening the file on linux and windows environments:

Emgu.CV.Util.CvException: OpenCV: !_src.empty()
   at Emgu.CV.CvInvoke.CvErrorHandler(Int32 status, IntPtr funcName, IntPtr errMsg, IntPtr fileName, Int32 line, IntPtr userData)
   at Emgu.CV.CvInvoke.cveCvtColor(IntPtr src, IntPtr dst, ColorConversion code, Int32 dstCn)
   at Emgu.CV.CvInvoke.CvtColor(IInputArray src, IOutputArray dst, ColorConversion code, Int32 dstCn)
   at UVtools.Core.FileFormats.FileFormat.SanitizeThumbnails(Boolean trimToFileSpec) in UVtools.Core\FileFormats\FileFormat.cs:line 4246
   at UVtools.Core.FileFormats.FileFormat.Decode(String fileFullPath, FileDecodeType fileDecodeType, OperationProgress progress) in UVtools.Core\FileFormats\FileFormat.cs:line 4578
   at UVtools.UI.MainWindow.<>c__DisplayClass97_0.<ProcessFile>b__2() in UVtools.UI\MainWindow.axaml.cs:line 1432

### How to reproduce

Open UVTools and import the file in the attached zip. A progress bar is shown of the loading, after which the error occurs. 

### Files

[test.zip](https://github.com/user-attachments/files/16574353/test.zip)

VasilNikolov, 2024-08-11:
Version 4.3.2 breaks too...

sn4k3, 2024-08-11:
Looks like all PNG's on your file are black/empty. Any reason why?

VasilNikolov, 2024-08-11:
Will investigate... this is the source stl.
[test.zip](https://github.com/user-attachments/files/16575542/test.zip)

sn4k3, 2024-08-11:
Your file is all messed up. Try to resclice.
The error is due all black layers and it try to generate thumbnail from empty layer.
[image]

VasilNikolov, 2024-08-12:
How can I open the files to check if there are any valid layers?

VasilNikolov, 2024-08-12:
I think it makes sense to not break the execution and still open the file in order to inspect it. Besides it has all the other properties from the slicer like time estimates, etc...

sn4k3, 2024-08-12:
It will be fixed on next patch.
To open the file you must do it manually, by open the .sl1 as ZIP and inspect files.

sn4k3, 2024-08-19:
Fixed

closed 2024-08-19 by sn4k3