sn4k3/UVtools #919: [Bug] Pixel arithmetic results in error: Emgu.CV.Util.CvException: OpenCV: 0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows

Drynwhyl, 2024-08-25:
### System

```Text
UVtools v4.4.0 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: AMD Ryzen 5 7600 6-Core Processor              
Processor cores: 12
Memory RAM: 22.81 / 27.62 GB
Runtime: win10-x64
Framework: .NET 6.0.32
AvaloniaUI: 11.1.1
OpenCV: 4.9.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: The Rotten Body_21h59m_119ml_copy1.pwma [Version: 516] [Class: PhotonWorkshopFile]
```

### Printer and Slicer

- Printer: Photon Mono 4K
- Slicer: Lychee 6.2.0

### Description of the bug

Following error occurs when I try to perform pixel arithmetic operation on multiple layers. It is not consistent, it may occur on different layers of the same file. If I'm lucky enough after 8-10 attempts it completes without error, but sometimes I just give up after 20+ tries:
Emgu.CV.Util.CvException: OpenCV: 0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows
   at Emgu.CV.CvInvoke.CvErrorHandler(Int32 status, IntPtr funcName, IntPtr errMsg, IntPtr fileName, Int32 line, IntPtr userData)
   at Emgu.CV.MatInvoke.cveMatCreateFromRect(IntPtr mat, Rectangle& roi)
   at UVtools.Core.Extensions.EmguExtensions.Roi(Mat mat, Rectangle roi, EmptyRoiBehaviour emptyRoiBehaviour) in UVtools.Core\Extensions\EmguExtensions.cs:line 944
   at UVtools.Core.Operations.OperationPixelArithmetic.<>c__DisplayClass149_0.<ExecuteInternally>b__0(Int64 layerIndex) in UVtools.Core\Operations\OperationPixelArithmetic.cs:line 924
   at System.Threading.Tasks.Parallel.<>c__DisplayClass20_0`1.<ForWorker64>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.<>c__DisplayClass20_0`1.<ForWorker64>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica.Execute()

### How to reproduce

1.  Go to "Tools" -> "Pixel Arithmethic"
2. Make sure layer range is set to all layers
3. Set Operator: Subtract with brightness
4. Set Apply to: Model surface
5. Set brightness: Use a pattern instead of fixed brightness
6. Set Alternate pattern every "1" layers
7. Set pattern to: 
0 32
32 0
8. Set alternate pattern to:
32 0
0 32
9. Click "Pixel arithmetic" button to start operation
10. Result: Error should occur

sn4k3, 2024-08-25:
This is the same as https://github.com/sn4k3/UVtools/issues/905 
I also cannot reproduce using your workflow.
Can you profile the file? Does it happen every time to you?

Drynwhyl, 2024-08-26:
It happens almost every time with large file (2000-3000 layers). If it's something small (around 300-500 layers) it can work first try.
Here is the file link: https://drive.google.com/file/d/1q6aH6RXziXLpUuh7nYFL_CXULw_LoCrW/view?usp=sharing

sn4k3, 2024-08-26:
Tested with your file and it works fine, I ran 5 times the operation without errors.
Need to dig into this and see if I can spot the problem

sn4k3, 2024-12-18:
Cant repo, if you get this again and able to reproduce at 100% the times please reopen.

closed 2024-12-18 by sn4k3