sn4k3/UVtools #924: [Bug] Exposure Time Finder - Mars 4 Ultra - Pattern Loaded Model vs Multiple Exposures

dominik-fr, 2024-08-30:
### System

```Text
UVtools v4.4.1 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: 12th Gen Intel(R) Core(TM) i5-12600KF
Processor cores: 16
Memory RAM: 16.73 / 31.86 GB
Runtime: win10-x64
Framework: .NET 6.0.33
AvaloniaUI: 11.1.3
OpenCV: 4.9.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1040    UA: 1920 x 1040
2: 2560 x 1440 @ 150%
    WA: 2560 x 1380    UA: 1707 x 920
3: 1080 x 1920 @ 100%
    WA: 1080 x 1880    UA: 1080 x 1880

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: xcxcxcxcxc.ctb [Version: 4] [Class: CTBEncryptedFile]
```

### Printer and Slicer

- Printer: Elagoo Mars 4 Ultra

### Description of the bug

Although Mars 4 Ultra can be setup with default Exposure Time Finder test model and adjusted with Multiple Exposures (see: https://www.youtube.com/shorts/0rYtb15FEDo) it is not possible to setup Multiple Exposures for custom, patter loaded, model. Probably more models might be affected by this behaviour.

### How to reproduce

1. Go to Exposure Time Finder
2. Setup any common properties/object properties
3. Setup Multiple Exposures -> this setup works on Mars 4 Ultra  (see: https://www.youtube.com/shorts/0rYtb15FEDo)
4. Enable Pattern Loaded model

Expected behaviour:
Same behaviour as for common model - multiple exposures are set up for custom model

Actual behaviour:
Multiple Exposures is disabled for enabled Pattern Loaded model
[image]

sn4k3, 2024-08-30:
Is this printer with the moving vat for peel layer?
Provide a sample file so I can check, something is triggering to disable that.
Under edit printer parameters if layer options are missing then thats the reason.

dominik-fr, 2024-08-31:
This is regular, move-plate-to-peel, printer. Nothing fancy. Same behaviour for .ctb and .goo.
[singleTestModel.zip](https://github.com/user-attachments/files/16824203/singleTestModel.zip)

After testing the thing a bit I've noticed that Multiple Exposures are getting disabled after this section is collapsed. If that section is left expanded one can add pattern model and setup various exposures as expected.

Before options collapsing:
[image]

After options collapsing:
[image]

sn4k3, 2024-08-31:
Good find, will be solved on next patch.

sn4k3, 2024-09-14:
Fixed

closed 2024-09-14 by sn4k3