sn4k3/UVtools #925: [Bug] Error while running detection pass

clear62, 2024-09-05:
### System

```Text
UVtools v4.4.1 X64
Operative system: Microsoft Windows 10.0.22631 X64
Processor: AMD Ryzen 7 5800H with Radeon Graphics         
Processor cores: 16
Memory RAM: 9.16 / 15.35 GB
Runtime: win10-x64
Framework: .NET 6.0.33
AvaloniaUI: 11.1.3
OpenCV: 4.9.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: ALL - 1 of each folder.goo [Version: 0] [Class: GooFile]
```

### Printer and Slicer

- Printer: Elegoo Saturn 3 Ultra
- Slicer: Lychee Slicer 6.0.200

### Description of the bug

While running a scan to detect issues in a print file, toward the end of the detection pass, the following error was generated.

System.AggregateException: One or more errors occurred. (External component has thrown an exception.)
 ---> System.Runtime.InteropServices.SEHException (0x80004005): External component has thrown an exception.
   at Emgu.CV.CvInvoke.cveConnectedComponentsWithStats(IntPtr image, IntPtr labels, IntPtr stats, IntPtr centroids, LineType connectivity, DepthType type, ConnectedComponentsAlgorithmsTypes cllType)
   at Emgu.CV.CvInvoke.ConnectedComponentsWithStats(IInputArray image, IOutputArray labels, IOutputArray stats, IOutputArray centroids, LineType connectivity, DepthType labelType, ConnectedComponentsAlgorithmsTypes cclType)
   at UVtools.Core.Managers.IssueManager.<>c__DisplayClass18_2.<DetectIssues>b__9(Int64 layerIndexInt) in UVtools.Core\Managers\IssueManager.cs:line 651
   at System.Threading.Tasks.Parallel.<>c__DisplayClass20_0`1.<ForWorker64>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.<>c__DisplayClass20_0`1.<ForWorker64>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica.Execute()
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.TaskReplicator.Run[TState](ReplicatableUserAction`1 action, ParallelOptions options, Boolean stopOnFirstFailure)
   at System.Threading.Tasks.Parallel.ForWorker64[TLocal](Int64 fromInclusive, Int64 toExclusive, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Func`4 bodyWithLocal, Func`1 localInit, Action`1 localFinally)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.ThrowSingleCancellationExceptionOrOtherException(ICollection exceptions, CancellationToken cancelToken, Exception otherException)
   at System.Threading.Tasks.Parallel.ForWorker64[TLocal](Int64 fromInclusive, Int64 toExclusive, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Func`4 bodyWithLocal, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.For(Int64 fromInclusive, Int64 toExclusive, ParallelOptions parallelOptions, Action`1 body)
   at UVtools.Core.Managers.IssueManager.DetectIssues(IssuesDetectionConfiguration config, OperationProgress progress) in UVtools.Core\Managers\IssueManager.cs:line 237
[image]

   at UVtools.UI.MainWindow.<>c__DisplayClass192_0.<ComputeIssues>b__0() in UVtools.UI\MainWindow.Issues.cs:line 609

### How to reproduce

Load .goo file into UV tools
activate pixel editor
activate difference tool
run a detect pass with default settings

### Files

https://wetransfer.com/downloads/9535b0d77407250995b4cdc827c515d920240905002533/f3aff86904dfcfafb35ebd8ed7a297e220240905002533/773901?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid

email link sent to self.  Hope it lets others download.  Never used wetransfer.com before

sn4k3, 2024-09-12:
I was unable to reproduce your problem. Does it happen everytime with the file?
I notice a bug on Difference tool but both pixel editor and difference does not link with issue detection.

sn4k3, 2024-09-12:
I think your problem is lack of RAM, at some point your file takes more RAM than you actually have. With a peak of 18GB of RAM in first 18s.
Your best bet is to upgrade RAM to 32GB or decrease core count in UVtools settings.

[image]

clear62, 2024-09-13:
Ok, I will attempt to make that change.  Thanks!

closed 2024-12-18 by sn4k3