sn4k3/UVtools #927: editable retraction settings and dual motion control for dynamic lifta

SPYKE357, 2024-09-06:
### Is your feature request related to a problem?

currently the dynamic lifts tool sets the retractions to a set speed from file settings, not enough control over this movement section

### Describe the solution you'd like

would be cool to have control over those movements. maybe a percentage split for distance to control speed down. eg for 75% of the movement back down i can set 180 and then for the remaining 25% i can set it to 60. maybe a dual motion/tsmc dynamic lift? is this even possible? would be really cool. currently setting the dynamic lift retract speed by making sure the speed i want to use is in the first box, eg 65mm/m but then im stuck with it for the full retraction

sn4k3, 2024-09-06:
Since this tool was made to work with one lift/retract makes no sense to have retract control. Retract have no peel force and we expect it always perform well, so we use a high speed and let acceleration do it job near the end of movement. Same goes for TSMC, it makes little sense to have dynamic retract speed. 
However a setting can be included to change that retract speeds if user want to change them together.

SPYKE357, 2024-09-09:
Hi Tiago. Thank you for the reply. yeh i think it would be cool to add the retract speed setting. or can it be set via the edit parameters tool? thank you again for this tool though. you have no idea how much it makes my halot mage pro at least tolerable. on a side note if i may, do you know which features in the tools and calibrations work with cxdlpv4. i tried the exposure time finder but the model came out very tall. i tried dynamic lifts but the printer doesnt seem to obey the speeds that are set. it seems to obey the lift heights though which is nice. any tips for the halot mage pro/cxdlpv4?

Força Tiago

sn4k3, 2024-09-09:
Retract settings can be adjusted on Edit parameters.
Myself I use "max" retract speed and let acceleration do it thing, but in case of a high accel you can use sane speeds, something like 150 to 200mm/min.

sn4k3, 2026-09-19:
Implemented in v7

closed 2026-09-19 by sn4k3