sn4k3/UVtools #941: Exposure Setting to Zero Seconds on Anycubic M5s Pro

Jewce86, 2024-11-19:
### System

```Text
UVtools v4.4.2 X64 and 4.4.3 X64
Operative system: Microsoft Windows 10.0.22631 X64
Processor: 12th Gen Intel(R) Core(TM) i9-12900K
Processor cores: 24
Memory RAM: 27.75 / 63.75 GB
Runtime: win10-x64
Framework: .NET 6.0.33
AvaloniaUI: 11.1.3
OpenCV: 4.9.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: 15Boyz()_03h21m_73ml_copy.m5sp [Version: 518] [Class: PhotonWorkshopFile]
```

### Printer and Slicer

- Printer: Anycubic M5s Pro v1.0.5.4
- Slicer: Anycubic Photon Workshop 3.4.2 & Chitubox V2.1

### Description of the bug

Exposure setting itself to 0s on printer for all layers following Bottom and Transition (including delay UV light activation during transition). When printing a file, the Bottom 5 layers work properly. When the transition layers start, there is a delay in activating the UV light. When the normal layers begin to print, rather than the UV light staying on for 3.6s, it flickers quickly. This can also be noticed in the slicer when the estimated time of the file drops significantly, in the linked file, original file time from slicer is 4h26m, in the UVTools processed file, with the issues of the exposure resetting to 0s, time is now 3h21m. Attempts to manually change exposure on the printer via the settings fail as it reverts back to 0s for the following layer. Attempts to print without processing in UVTools works successfully. Prints immediately fail after processing through UVTools as there is no signficant exposure.

This was tested and confirmed by turning off all AI settings such as Resin Detect and Intellegent Release, covering the screen with a sheet of paper to protect it from the build plate, and watching the UVLight activating via a side hole where the Vat would sit.

Another issue i had noticed in UV Tools 4.4.3; files would take upwards of 15 minutes to Detect Issues. When i downgraded to UVTools 4.4.2, files returned to the normal approximately 3 minutes. When printing with my previous Mono 2; these issues were not happening, as well as the Computing Issues were only taking 3 minutes.

### How to reproduce

Slice File in Anycubic Slicer
Save File to Disk
Open file in UVTools
Compute Issues
Fix Resin Traps
Save File
Load file on printer via either USB or Cloud (tested both)
Print file.

This was tested and confirmed by turning off all AI settings such as Resin Detect and Intellegent Release, covering the screen with a sheet of paper to protect it from the build plate, and watching the UVLight activating via a side hole where the Vat would sit.

### Files

[https://1drv.ms/u/c/d9d92a66e33001da/EUyDkmGFYeZGlKS0E3R2wroBB7wVBva_RF_PC9WfV6MKIg?e=cgA1CD](url)

sn4k3, 2024-12-03:
Looks like `PerLayerOverride` is activated which cause problem in recent models.
In same file you can try to: `File - Reset layer properties`, Save and try to print again. My bet is that will work as that reset all properties and set flag to off.

Jewce86, 2024-12-30:
Sorry for not seeing this sooner. I've tried it and it doesn't seem to be working. I do see the function you are referring to; when i bring the file into UVTools, it shows PerLayerOverride as FALSE. I fill/solidify resin traps, then go to SAVE. As soon as I click save, PerLayerOverride goes to TRUE. I tried using RESET LAYER PROPERTIES, but it continues to remain as TRUE and doesn't reset to FALSE. Is there another way to modify PerLayerOverride?

sn4k3, 2024-12-30:
Did you try with latest UVtools release? It have some fixes regarding that.

Jewce86, 2025-01-03:
Yes, UVTools 5.0.3, as well as i have the most up to date version of Anycubic Slicer, 3.5.0_20241210140631. It's the exact same issue i'm having even when I export directly from Chitubox; it's like my machine does not like printing files from anything other than stuff saved from Anycubic Slicer; once i get past the transition layers, exposure time sets to 0.0s. I'm printing the exact same file now, which was sliced in Anycubic without post-processing in UVTools, and it's showing my exposure a 3.6s

sn4k3, 2025-01-03:
Can you provide a file sample with that problem?

JoveToo, 2025-01-04:
Have you saved after resetting the layer properties? PerLayerOverride only goes to False if you Save...

Jewce86, 2025-01-05:
> Can you provide a file sample with that problem?

Here's the what I tried to print. This one was exported from Anycubic. PerLayerOverride shows as False when working with Files done in Anycubic. This file should be over 2.5hrs, but shows as 1h50m and automatically sets my layer exposure to 0.0s as soon as the bottom layers are done. 

https://drive.google.com/file/d/1II8tVMg_tDcwHNBKUW5jEtX_uihj39rB/view?usp=sharing

Jewce86, 2025-01-05:
> Have you saved after resetting the layer properties? PerLayerOverride only goes to False if you Save...

Yes, when i export from Chitubox, reset layer properties and save, it switches from FALSE to TRUE, but i'm noticing Chitubox is having a whole slew of issues as it does the 0.0s exposure without even post-processing it in UVTools

sn4k3, 2025-01-07:
I tried your file, PerLayerOverride is `False`, if I save it remains as `False`.
That way it can only be any setting you are applyng

Jewce86, 2025-01-16:
I get that, but when I print the, the m5s Pro reads the file as 0.0s exposure. No issues exporting directly from Anycubic, so it's not the printer, it's thr saved from from UVTools. Not sure why the printer is reading the settings as 0.0s

K4rdis, 2025-01-29:
I have same issue with my M5s, UVTools 5.0.5.

genesi5, 2025-03-25:
Can confirm, had the same issue with 5.0.3 (i guess), Photon Mono M5s. 
Opening .pm5s file with Anycubic slicer and then sending it to printer helped to recover initial exposure time

sn4k3, 2025-06-14:
> This started to suddenly happen on my M5S Pro after 1 year of using it with no issues.

Did you upgrade firmware recently?

Hargrim, 2025-07-12:
@sn4k3 I did some more prints after downgrading the firmware and it started to happen randomly from time to time. I have an example of file before I modified it with UVTools and one that sets normal layer exposure time to 0.0 - could I somehow share those with you so you could have a look what's different?

sn4k3, 2025-07-12:
You can zip your files and send here or to my email. 
Use a service like wetransfer or other to get the download link

sn4k3, 2025-07-13:
```
PS D:\Tiago\Desktop> UVtoolsCmd.exe compare '.\Westfalia Trench 2 - file broken via UVTools.m5sp' '.\Westfalia Trench 2 - Original file, not modified.m5sp' --without-layers
Opening file Westfalia Trench 2 - file broken via UVTools.m5sp: Done in 0.23s
Opening file Westfalia Trench 2 - Original file, not modified.m5sp: Done in 0.15s
[Global #19]
LayerCount: 2894 != 2895
PrintTime: 31247.35 != 30346.84
MaterialCost: 111.47 != 111.344
LastLayerIndex: 2893 != 2894
AllLayersAreUsingGlobalParameters: True != False
UsingPerLayerSettings: False != True
RequireFullEncode: False != True
PrintHeight: 86.82 != 86.85
Count: 2894 != 2895
MaximumPossibleTransitionLayerCount: 2888 != 2889
NormalLayerCount: 2889 != 2890
PrintTimeComputed: 31247.35 != 30346.84
PrintTimeHours: 8.68 != 8.43
PrintTimeString: 08h40m != 08h25m
DisplayTotalOnTime: 6819.53 != 6821.83
DisplayTotalOnTimeString: 01h53m39s != 01h53m41s
DisplayTotalOffTime: 24427.82 != 23525.01
DisplayTotalOffTimeString: 06h47m07s != 06h32m05s
StartingMaterialCost: 111.47 != 111.344
```

File A have less 1 layer than B but the real problem is here: `UsingPerLayerSettings`
As stated in this issue that flag is causing all the problems, you can reset to false by going at File -> Reset layer properties, make sure to hold Shift when clicking to perform a full reset

sn4k3, 2025-07-13:
I will send you the original file just with flag to true and nothing more, for you to test if that alone is what cause the problem