sn4k3/UVtools #953: [FEATURE REQUEST]  Anyway to calculate how much RAM you need ?

MasterCATZ, 2024-12-13:
### Is your feature request related to a problem?

I have 128gb RAM and kept getting OOM ,
within minutes UVtools would be  over 90gb  Used and had to close down everything else for UVtools to not crash my desktop 

Ubuntu user , multiple times UVTools would just cause entire desktop to reload , (while hollowing / adding infill)
 can their be  some sort of guard against this ? why can the program not kill its self ?
using UVtools.AppImage 

Currently deciding with going 512gb 3200mhz RAM and 1tb 2400mhz RAM 

as much as I would love having 1Tb  for it to be cost effective I would be taking a 28% performance hit 
I rather not spend more than $1500 on more memory 
if I knew 512gb would not have a chance of OOM'ing  I would rather go that route so my data scrubs are not slowed down 
but if UVtools is going to be another memory hog contender with chrome I guess i will jut have to increase the budget 

so is their some calculation that can be done based on layers and dimensional size to work out memory usage ? 

I was using working with ~250x250x250mm  @50um  and I can do ~900x900x300mm @25um 

### Describe the solution you'd like

Program to kill itsself if a set memory usage threshold is reached

sn4k3, 2024-12-16:
You have a ton of cores, is that 128 the total or it doubles with hyperthreading?
Note that by default UVtools will spawn a task per core, meanwhile if task is late or there are free tasks it will spawn a bit more to compensate. That of course require a lot of ram. Most of operations will take more than a bitmap, sometime three or more bitmap in issue detection. Some memory are only free latter (Not just after task ends).
If by any chance there are a memory leak your system will be killed easy.

My advice is to cut workload in half on UVtools settings, start with 1/2, monitor ram and tune from there.

You can also attach a file here and tell the workflow so I can check it.
Infill operation does take alot of ram...

> UVtools is going to be another memory hog contender with chrome 

Chrome has no chance here 😁

sn4k3, 2024-12-18:
Implemented a limiter, please test it.

Hargrim, 2024-12-30:
> Implemented a limiter, please test it.

I'm not the OP, but I have an issue with this feature. I have 32GB of ram.

The limiter is killing my UVTools app when trying to drill through multiple selected Suction Cups with below error:

[image]

It is working if I click suction cups one by one and drill them, but selecting 10 at once and clicking drill on all of them crashes the app.

Would it be possible to add 3rd dropdown option to settings that would restore how UVTools was working a few version back? Like 'do nothing' option in the dropdown? Currently there is only Pause and Cancel.

[image]

Text of the error:
```
System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (Your system memory RAM hit the limit of 31.7GB from a total of 31.93GB.  Available: 0.23GB.  UVtools: 14.82GB (46.41%).
The program crashed on purpose due the impossibility to pause or cancel the running operation, this was to relief pressure and ensure the system stability.

Monitor your memory RAM under the system tools and re-run the operation to check the usages and find the cause.
You may need to stop other processes or increase your memory RAM in order to deal with huge files and/or heavy operations.

Currently you have configured a lower limit of 1GB of available memory RAM for the operations.)
 ---> System.InsufficientMemoryException: Your system memory RAM hit the limit of 31.7GB from a total of 31.93GB.  Available: 0.23GB.  UVtools: 14.82GB (46.41%).
The program crashed on purpose due the impossibility to pause or cancel the running operation, this was to relief pressure and ensure the system stability.

Monitor your memory RAM under the system tools and re-run the operation to check the usages and find the cause.
You may need to stop other processes or increase your memory RAM in order to deal with huge files and/or heavy operations.

Currently you have configured a lower limit of 1GB of available memory RAM for the operations.
   at UVtools.UI.MainWindow.<>c__DisplayClass57_0.<RamUsageTimerOnElapsed>b__0() in UVtools.UI\MainWindow.axaml.cs:line 518
   at Avalonia.Threading.DispatcherOperation.InvokeCore()
   --- End of inner exception stack trace ---
Category: Task
File: Magori Ostland DMS MK3 2 — kopia.m5sp  [Version: 518] [Class: AnycubicFile]

Machine date time: 30.12.2024 20:15:23
    UTC date time: 30.12.2024 19:15:2
```

sn4k3, 2024-12-30:
@Hargrim This is the excepted and working as it should, the limiter will kill UVtools to prevent system crash as is explained on the text:

> The program crashed on purpose due the impossibility to pause or cancel the running operation, this was to relief pressure and ensure the system stability.

This happens to all operations that can't be cancelled or paused. Drill is one of them. I can however try to implement an cancel on that operation. Drill also takes a ton of ram because it need to get many images to scan for pixels up to a void.

Hargrim, 2024-12-31:
Thanks for reply and continuous work on this!

Would it be possible to disable this check from the settings? Or disable it for drilling only via checkbox or something similar? 

It was working fine for me before this update, even if it was taking all RAM on my PC. Now I need to right click each suction cup and drill them one by one which is a downgrade and lose of functionality from my POV. Adding cancel option will not improve the situation, I will still need to right click one suction cup after another to drill them, now with added cancel, which will change nothing - if I understand correctly what you are suggesting above.

sn4k3, 2024-12-31:
Just like any other feature, you can set it to 0 to disable it.

Hargrim, 2024-12-31:
Setting it to 0 helped, thanks!

MasterCATZ, 2025-05-29:
sorry for not replying I only got a notification when it was closed 

256 cores 512Gb is what it is currently running  with  AMD EPYC 7C13 64-Core Processor's 
I did end up finding 1/4 cores seemed to work , that also allowed all cores to be 3.1ghz 

thanks for adding the limiter function anyhow at least that saves the desktop crashing 

what I was really after was something that would show what the max cores would be allowed for parallelism or a way to set max memory usage for program to use , 
pc idle after boot is about 64gb and chrome bumps that to 128gb  ..  leaving 300gb for UVTools 

instead of having a core limiter 
if their could be a calculator that can have MAX RAM Usage set then calculation used to spit out the MAX core count to use based on the enabled RAM compression ?

thanks for creating a fantastic program and still supporting it , are you even doing resin printing yet ?
from memory you made this without even having a printer

sn4k3, 2025-05-29:
> how much time was spent Implementing the limiter ? ,

[on Dec 18, 2024](https://github.com/sn4k3/UVtools/issues/953#issuecomment-2552354203)

> if their could be a calculator 

It could, but that really depends on your file, not all files have same resolution and that is a big variable in play. Also system memory management not all the same. The best bet is test and keep a safe margin, those mega core CPUs will make UVtools eat ram at will.

> are you even doing resin printing yet ?
> from memory you made this without even having a printer

I brought a IBEE and hack it to open source klipper.
But i dislike resin, rarely print anything other than a covid model from time to time.

sn4k3, 2025-05-30:
By now I can't remember, maybe a few hours to a working day. All the limiter does is pause the operation when hitting certain amount of free ram, if unable to pause it closes the app as safe procedure.

> especially as it then created negative side affects for other users

I believe that is pausing many operations since files grow in resolution, and even if you have a huge amount of RAM with high core count it fast deplete ram. Windows have better SWAP management than linux and there weren't any report of bad behavior on windows because lack of ram.

I think GPU would never take a place in UVtools, the program is strongly bound to CPU, the cost of GPU - CPU transfer kills all performance. It requires sequential and continuous work on GPU. Someday if openCV made all functions compatible with GPU (I doubt)... 
Anyway most people have better CPU+RAM combo than dGPU. The core count of GPU is high but RAM is on short side if you having trouble with 128GB imagine with 4GB and 8GB GPUs...

> again thanks for supporting something you still not not even use
I wish i never started resin printing the house is full off VOC's from it constantly feels like my lungs are on fire

I do FDM, but mSLA I have no patience to all care you must have and post process thing, it just don't worth.
I made the project for fun and to revive my C# memories, in a hope that I could use PS when I get my printer.
Now I'm about to release other project, (network nature).
A piece of advice, please get that printer away from you and house habitants, people have no clue the bad this technology causes to them and planet overall! You can develop breathing problem if you keep cohabit with active printer. I even go beyond saying these should not permitted to anyone, but just a restricted / regulated / licensed people. Even FDM causes a massive harm, the emissions of VOCs and tons of microplastics floating away and come into our lungs are no joke. People are not aware because they can't see, but it's an airborne harzard. The fact that anyone can own a printer and easy access to nasty chemical pollutants is what I advocate against, and the truth is most people are just printing toys and try to making money selling trash, >95% of the prints are just trash, no real use, no application, just waste. 3D printers as they are used by most are just weapons against the planet. But this is also true for everything else, where we (Humans) put hands in technology we always use them badly and for the worse.
Now schools have 3D printers, and there is not a single class to educate about proper use, mindset, the risks, how to be handled, nothing! Just let print a toy and breath in the tasty smell from "biodegradable" PLA! The fact is that educators are also unaware.