sn4k3/UVtools #956: [Bug] AvaloniaControls.AdvancedImageBox Pixel Grid lines dont connect when image is smaller than control size

AITYunivers, 2024-12-18:
### System

```Text
As AvaloniaControls is not UVTools (the program), I have replaced some data to be relevant

UVtools.AvaloniaControls v4.0.0 X64
Operative system: Microsoft Windows 10.0.22631 X64 (Windows 11)
Processor: AMD Ryzen 5 3600 6-Core Processor              
Processor cores: 12
Memory RAM: 20.19 / 63.91 GB
Runtime: win-x64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.2

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1032    UA: 1920 x 1032
```

### Description of the bug

When using a small image on the AdvancedImageBox, and then scaling that image up to the point where all 4 edges are still visible, along with the Pixel Grid, the lines in the Pixel Grid are misaligned to the image. The issue goes away when zoomed in.

### How to reproduce

1. Create a new Avalonia 11.2.2 project
2. Import UVtools.AvaloniaControls 4.0.0 from Nuget
3. In your MainWindow, add
`xmlns:uv="clr-namespace:UVtools.AvaloniaControls;assembly=UVtools.AvaloniaControls"`
4. Set the content of your MainWindow to
`<uv:AdvancedImageBox AutoZoomToFit="True"/>`
5. Set the `Image` property of the AdvancedImageBox to a small image, in my case I used a 32x32 image

### Files

[image]

sn4k3, 2024-12-18:
It feels like you have been waiting for the v4 release because I just post it... 😅

I have tested and I observe the same, it's strange, I will check it out...
[image]

AITYunivers, 2024-12-18:
Nope just really coincidental timing, I found this yesterday, and when I downloaded it, 4.0.0 came out within the same 10 seconds.

sn4k3, 2024-12-18:
Should be fixed now, try update and report back.

[image]

AITYunivers, 2024-12-18:
Looks good, thanks!
[image]

closed 2024-12-18 by AITYunivers