sn4k3/UVtools #957: [Bug]

Gruntsky, 2024-12-19:
### System

```Text
UVtools v5.0.0 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: AMD Ryzen 5 3600 6-Core Processor              
Processor cores: 12
Memory RAM: 10.71 / 31.89 GB
Runtime: win-x64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.2
OpenCV: 4.9.0

Path:       E:\UVTools\
Executable: E:\UVTools\UVtools.exe
Loaded file: BackUV.goo [Version: 0] [Class: GooFile]
```

### Printer and Slicer

- Printer: Saturn 4 Ultra
- Slicer: Lychee

### Description of the bug

Updated to newest version, now when I try to save an edited file I receive this error:
[image]

### How to reproduce

1. Go to file.
2. Click Save

mjf55, 2024-12-19:
I have the same issue.   installed V5 and when I go to launch UVTools, I get the same crash report as Gruntsky

sn4k3, 2024-12-19:
I get an error while saving goo file but is different from yours.
On your error I can see that you are opening an SL1 file and not saving a goo file.
If you you can please send me the problematic file.

Also try to replace this file on instalation folder and report back: [UVtools.Core.zip](https://github.com/user-attachments/files/18192294/UVtools.Core.zip)

mjf55, 2024-12-19:
I tried your UVTools-core dll my my issue, but no change.  My issue shows on launch.  Do you want me to create a different issue?

sn4k3, 2024-12-19:
> Do you want me to create a different issue?

No, it may be related to the OP, as the launch it open the SL1 file which is a ZIP.
I tried to open UVtools with demo file but works fine on my PC.

Let me find some information about this. Meanwhile revert back to previous version.

sn4k3, 2024-12-19:
Looks like the problem is the missing `System.IO.Compression.Native.dll` which I have no my PC but is not on UVtools folder, somehow it was not copied by the compiler... Need to find why.

Gruntsky, 2024-12-19:
Just replaced the .dll file and opened the file. Save looks to have worked with the new .dll

sn4k3, 2024-12-19:
@mjf55 I just downloaded the v5 zip and Native.dll is there... I also reinstalled via MSI and I got the native dll in my folder.
Can you confirm the presence of `System.IO.Compression.Native.dll` on your installation folder?

Gruntsky, 2024-12-19:
Yes, I can confirm System.IO.Compression.Native.dll is present in installation folder
[image]

sn4k3, 2024-12-19:
Found the cause: https://github.com/dotnet/runtime/issues/82518
NET 6 use this dll as v42, that was a wrong version which they fixed on NET7, meaning MSI won't upgrade the file because version is lower. @Gruntsky have the new Native.dll because he reinstalled UVtools which cleans and put all files without checking for upgrade.

For who land here, Uninstall UVtools and Reinstall.

closed 2024-12-19 by Gruntsky