sn4k3/UVtools #959: [FEATURE REQUEST] Add functionality to the Padding property in AdvancedImageBox

AITYunivers, 2024-12-19:
### Is your feature request related to a problem?

I'd like to have a padding around my image that I can still pan over to, like how in Paint.net, the view area is 3360x1550 on a 1x1 image, but the control does not have any equivalent features as far as I could tell.

### Describe the solution you'd like

I would like the Padding property to be used in a way where you can still pan the view around around the image rather than exclusively within the image bounds.

### Files

Paint.net for reference:
[image]

sn4k3, 2024-12-19:
I have padding working with other fill size methods, however for zoom mode is a bit more complicated.
The video you show is not just a padding, I think they call "Infinite canvas" to that.
I will take a deeper look

sn4k3, 2026-09-19:
Implemented, please check `PanBoundsMode` and set `Padding` to like `500`, report back

AITYunivers, 2026-09-19:
I'll dig up this old project just to test this out for ya lol, thanks for your hard work!

AITYunivers, 2026-09-19:
Wow I completely forgot how much I hate Avalonia. Couldn't get the original project working, but tested it in a new app.
Really cool, does what I wanted. Seems to have respawned #956 in a way though.

[image]

sn4k3, 2026-09-19:
Thanks, I will check. Grids now are reused tiles instead of drawing lines, maybe that respawn the problem.

sn4k3, 2026-09-19:
On this side looks ok: [image]

AITYunivers, 2026-09-19:
Here's the axaml I used
```xml
<Window xmlns="https://github.com/avaloniaui"
        xmlns:x="http://schemas.microsoft.com/winfx/2006/xaml"
        xmlns:d="http://schemas.microsoft.com/expression/blend/2008"
        xmlns:mc="http://schemas.openxmlformats.org/markup-compatibility/2006"
        xmlns:uv="clr-namespace:UVtools.AvaloniaControls;assembly=UVtools.AvaloniaControls"
        mc:Ignorable="d" d:DesignWidth="800" d:DesignHeight="450"
        x:Class="AvaloniaApplication1.MainWindow"
        Title="AvaloniaApplication1">
	<uv:AdvancedImageBox Grid.Column="2" PanBoundsMode="Padding" Padding="500" Image="MMFActive.png" AutoZoomToFit="True" AutoCenter="True" ConstrainZoomOutToFitLevel="False" />
</Window>
```

sn4k3, 2026-09-19:
With your xml I got:

[image]

AITYunivers, 2026-09-19:
it only happened near the bottom of the screen

sn4k3, 2026-10-09:
I think its fixed now

closed 2026-10-09 by sn4k3