sn4k3/UVtools #961: [Bug] Exception when saving .pwmb files

JRouLar, 2024-12-19:
### System

```Text
UVtools v5.0.0 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Core(TM) i7-14700K
Processor cores: 28
Memory RAM: 13.24 / 63.78 GB
Runtime: win-x64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.2
OpenCV: 4.9.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: Assault x50 - ACTough - uvtools.pwmb [Version: 516] [Class: AnycubicFile]
```

### Printer and Slicer

- Printer: Anycubic Photon Mono 6K
- Slicer: Lychee

### Description of the bug

When saving a .pwmb file, I get the following exception:

System.ArgumentOutOfRangeException: Specified argument was out of the range of valid values.
   at UVtools.Core.FileFormats.FileFormat.EncodeImage(String dataType, Mat mat) in UVtools.Core\FileFormats\FileFormat.cs:line 846
   at UVtools.Core.FileFormats.AnycubicFile.Preview..ctor(Mat mat) in UVtools.Core\FileFormats\AnycubicFile.cs:line 447
   at UVtools.Core.FileFormats.AnycubicFile.EncodeInternally(OperationProgress progress) in UVtools.Core\FileFormats\AnycubicFile.cs:line 1843
   at UVtools.Core.FileFormats.FileFormat.Encode(String fileFullPath, OperationProgress progress) in UVtools.Core\FileFormats\FileFormat.cs:line 4474
   at UVtools.UI.MainWindow.<>c__DisplayClass105_0.<SaveFile>b__2() in UVtools.UI\MainWindow.axaml.cs:line 2273

I just updated UVTools today. Previous version didn't fail, but my printer couldn't read the pwmb files.

### How to reproduce

Open the attached file.
Fill all the resin traps.
Save As (I used the existing name, plus -uvtools)
Exception...

### Files

[Assault x50 - ACTough - 1h33m.zip](https://github.com/user-attachments/files/18202478/Assault.x50.-.ACTough.-.1h33m.zip)

sn4k3, 2024-12-19:
That was fixed a version after, please update to latest version. You still on v5.0.0...

JRouLar, 2024-12-19:
> That was fixed a version after, please update to latest version. You still on v5.0.0...

Indeed, I can see a v5.0.2.

closed 2024-12-19 by JRouLar