sn4k3/UVtools #964: [Crash] System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (org.freedesktop.DBus.Error.ServiceUnknown: The name com.canonical.AppMenu.Registrar was not provided by any .service files)

nARN, 2024-12-20:
### System

```Text
UVtools v5.0.2 X64
Operative system: Ubuntu 24.04.1 LTS X64
Processor: Intel(R) Core(TM) i9-9900K CPU @ 3.60GHz
Processor cores: 16
Memory RAM: 20.79 / 125.73 GB
Runtime: linux-x64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.3
OpenCV: 4.9.0

Path:       /tmp/.mount_uvtoolV1z0pR/usr/bin/
Executable: /tmp/.mount_uvtoolV1z0pR/usr/bin/UVtools
```

### Description of the bug

```
System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (org.freedesktop.DBus.Error.ServiceUnknown: The name com.canonical.AppMenu.Registrar was not provided by any .service files)
 ---> Tmds.DBus.Protocol.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name com.canonical.AppMenu.Registrar was not provided by any .service files
   at Tmds.DBus.Protocol.DBusConnection.MyValueTaskSource`1.System.Threading.Tasks.Sources.IValueTaskSource.GetResult(Int16 token)
   at Tmds.DBus.Protocol.DBusConnection.CallMethodAsync(MessageBuffer message)
   at Tmds.DBus.Protocol.Connection.CallMethodAsync(MessageBuffer message)
   --- End of inner exception stack trace ---
Category: Task
File: 1200prc_0.05mm.encrypted.ctb  [Version: 5] [Class: CTBEncryptedFile]

Machine date time: 20/12/2024 14:51:51
    UTC date time: 20/12/2024 14:51:51
```

### How to reproduce

I struggle to reproduce it, so suspect it may be some sort of a race. When I got this exception I had just reloaded a file and went straight into "Edit print parameters". It works fine now, but on DBus I can still see calls to `com.canonical.AppMenu.Registrar` and errors in response:
```
method call time=1734709363.187768 sender=:1.302 -> destination=com.canonical.AppMenu.Registrar serial=295 path=/com/canonical/AppMenu/Registrar; interface=com.canonical.AppMenu.Registrar; member=RegisterWindow
   uint32 41943155
   object path "/net/avaloniaui/dbusmenu/1032ada03730481484f5810ae4363f30"
error time=1734709363.187791 sender=org.freedesktop.DBus -> destination=:1.302 error_name=org.freedesktop.DBus.Error.ServiceUnknown reply_serial=295
   string "The name com.canonical.AppMenu.Registrar was not provided by any .service files"
```

so I'm suspecting that the problem is that for some reason the exception was not caught at that time.

sn4k3, 2024-12-20:
This comes from UI Framework, and other more have this problem: https://github.com/sn4k3/UVtools/issues?q=DBus
I don't know how to resolve it as it come from external source.
I'm gona ask to Avalonia...

diefans, 2024-12-22:
for me this errors occurs since I use version 5.0.2...

sn4k3, 2024-12-22:
Because v5 uses latest avalonia. If you revert to latest v4 it won't trigger this error.

JoveToo, 2025-01-04:
It's really not rare with UVTools.

Eisenknochen, 2025-01-26:
same  problem here (Using Linux Mint)

sn4k3, 2025-02-22:
Is this fixed with latest release?

JKBert, 2025-03-05:
Nope. I've just tested (Linux Devuan).

sn4k3, 2025-03-06:
Dam, I tested on VM and works ok. Ubuntu as OP using. 
What is the most easy way to reproduce this?

DodoidOfficial, 2025-03-09:
I can confirm I am also running into this on Debian Testing (Trixie) w/ KDE. What seems to trigger it most consistently is to attempt to hover over the stats in the bottom bar (e.g. Wait Time). The tooltip will flash up a couple of times (but is seemingly empty?), and UVtools usually crashes after a few seconds of this.

sn4k3, 2025-03-10:
Check last version

DodoidOfficial, 2025-03-10:
This now appears to work for me, both in the release version and building from source. The contents of the tooltip still are not visible, and the tooltip still disappears immediately after appearing, but it no longer seems to crash for me.

sn4k3, 2025-03-10:
Yes, bug still there, is just ignored

https://github.com/user-attachments/assets/9cdcb6a8-8126-4088-a424-6b2e953120e3

JKBert, 2025-03-11:
Tiago Conceição a écrit :
> Yes, bug still there, is just ignored
>
> https://github.com/user-attachments/assets/9cdcb6a8-8126-4088-a424-6b2e953120e3

	I have rebuilt UVtools from scratch this morning. 5.0.8 crashes as
older releases with the same error.

	Best regards,

	JB

sn4k3, 2025-03-11:
> I have rebuilt UVtools from scratch this morning. 5.0.8 crashes as
> older releases with the same error.
> 
> 	Best regards,
> 
> 	JB

Can you post the error? For it to show it must be slight different, as I'm doing a message check

JKBert, 2025-03-23:
> > I have rebuilt UVtools from scratch this morning. 5.0.8 crashes as
> > older releases with the same error.
> > Best regards,
> > JB
> 
> Can you post the error? For it to show it must be slight different, as I'm doing a message check

Hello,

I have done some tests. Here is a "standard" error message:
```
System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (org.freedesktop.DBus.Error.UnknownMethod: Object does not exist at path “/com/canonical/AppMenu/Registrar”)
 ---> Tmds.DBus.Protocol.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Object does not exist at path “/com/canonical/AppMenu/Registrar”
   at Tmds.DBus.Protocol.DBusConnection.MyValueTaskSource`1.System.Threading.Tasks.Sources.IValueTaskSource.GetResult(Int16 token)
   at Tmds.DBus.Protocol.DBusConnection.CallMethodAsync(MessageBuffer message)
   at Tmds.DBus.Protocol.Connection.CallMethodAsync(MessageBuffer message)
   --- End of inner exception stack trace ---
Category: Task
File: UVtools_demo_file.sl1s  [Version: 0] [Class: SL1File]

Machine date time: 23/03/2025 14:08:41
    UTC date time: 23/03/2025 13:08:41
```
All information about configuration:
```
UVtools v5.0.8 X64
Operative system: Devuan GNU/Linux 6 (excalibur/ceres) X64
Processor: Intel(R) Core(TM) i9-10900F CPU @ 2.80GHz
Processor cores: 20
Memory RAM: 21.34 / 31.25 GB
Runtime: linux-x64
Framework: .NET 9.0.2
AvaloniaUI: 11.2.5
OpenCV: 4.10.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1080    UA: 1920 x 1080
2: 1920 x 1080 @ 100%
    WA: 1920 x 1080    UA: 1920 x 1080

Path:       /home/bertrand/git/UVtools/artifacts/publish/UVtools_linux-x64_v5.0.8/
Executable: /home/bertrand/git/UVtools/artifacts/publish/UVtools_linux-x64_v5.0.8/UVtools

General configuration for OpenCV 4.10.0 =====================================
  Version control:               3.4.2-10289-g3beff6e6c5

  Platform:
    Timestamp:                   2025-01-18T05:18:45Z
    Host:                        Linux 6.8.0-51-generic x86_64
    CMake:                       3.28.3
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/gmake
    Configuration:               Release

  CPU/HW features:
    Baseline:                    SSE SSE2 SSE3
      requested:                 SSE3
    Dispatched code generation:  SSE4_1 SSE4_2 FP16 AVX AVX2 AVX512_SKX
      requested:                 SSE4_1 SSE4_2 AVX FP16 AVX2 AVX512_SKX
      SSE4_1 (13 files):         + SSSE3 SSE4_1
      SSE4_2 (1 files):          + SSSE3 SSE4_1 POPCNT SSE4_2
      FP16 (0 files):            + SSSE3 SSE4_1 POPCNT SSE4_2 FP16 AVX
      AVX (3 files):             + SSSE3 SSE4_1 POPCNT SSE4_2 AVX
      AVX2 (25 files):           + SSSE3 SSE4_1 POPCNT SSE4_2 FP16 FMA3 AVX AVX2
      AVX512_SKX (2 files):      + SSSE3 SSE4_1 POPCNT SSE4_2 FP16 FMA3 AVX AVX2 AVX_512F AVX512_COMMON AVX512_SKX

  C/C++:
    Built as dynamic libs?:      NO
    C++ standard:                17
[... 132 lines cut ...]
64: System.Text.RegularExpressions, Version=9.0.0.0
65: HarfBuzzSharp, Version=1.0.0.0
66: System.Diagnostics.Process, Version=9.0.0.0
67: System.IO.Pipes, Version=9.0.0.0
68: Emgu.CV, Version=4.10.0.5680
69: System.Web.HttpUtility, Version=9.0.0.0
70: System.Text.Json, Version=9.0.0.0
71: System.Text.Encodings.Web, Version=9.0.0.0
72: System.Runtime.Intrinsics, Version=9.0.0.0
73: Markdown.Avalonia, Version=11.0.2.0
74: BinarySerializer, Version=8.6.4.1
75: System.Threading.Tasks, Version=9.0.0.0
76: System.IO, Version=9.0.0.0
77: System.Text.Encoding, Version=9.0.0.0
78: System.Reflection.TypeExtensions, Version=9.0.0.0
79: System.Reflection, Version=9.0.0.0
80: System.Reflection.Extensions, Version=9.0.0.0
81: System.Runtime.InteropServices.RuntimeInformation, Version=9.0.0.0
82: System.Linq.Expressions, Version=9.0.0.0
83: System.Reflection.Emit.Lightweight, Version=9.0.0.0
84: Anonymously Hosted DynamicMethods Assembly, Version=0.0.0.0
85: System.Runtime.Extensions, Version=9.0.0.0
86: System.ComponentModel.Annotations, Version=9.0.0.0

```

sn4k3, 2025-03-23:
@JKBert The error you getting is not included in my bypass, your is `UnknownMethod` while the current bypass is `ServiceUnknown`.
I can include both in next release.

JKBert, 2025-03-25:
I have added a quick and dirty patch to ignore this error in UVtools.UI/Program.cs and UVtools now runs as expected.

```
    private static void HandleUnhandledException(string category, Exception ex)
    { 
        ErrorLog.AppendLine($"Fatal {category} Error", ex.ToString());

        if (category == "Task")
        {
            if (ex.Message.Contains("org.freedesktop.DBus.Error.ServiceUnknown")) return; 
            if (ex.Message.Contains("org.freedesktop.DBus.Error.UnknownMethod")) return;
        }
```