sn4k3/UVtools #966: [Bug]

bvfhur, 2024-12-20:
### System

```Text
UVtools v5.0.2 X64
Operative system: Microsoft Windows 10.0.26100 X64
Graphic card: 
Processor: Intel(R) Core(TM) i5-1035G1 CPU @ 1.00GHz
Processor cores: 8
Memory RAM: 6.67 / 11.75 GB
Runtime: win-x64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.3
OpenCV: 4.9.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: Crab20a20presupported.stl_0.05_3.2_2024_12_20_12_40_00_03h25m_4ml_copy.cxdlp [Version: 3] [Class: CrealityCXDLPFile]
```

### Printer and Slicer

- Printer: 
- Slicer: 

### Description of the bug

pixel editor in line push format is off by2 pixels

### How to reproduce

1 go to the pixel editor
2 select line brush shape
3 "2" bush size

sn4k3, 2024-12-20:
You mean compared to the preview?
In drawing it does exactly two pixels for me, as well as 3 and 4 does exactly that pixel count.

However I know openCV line is not perfect, some times it does more or less 1 pixel than selected, this is a issue from openCV that I can't resolve. Currently UVtools does some kind of compensation but not always match the selected pixels count

bvfhur, 2024-12-20:
yes just compared to the preview its off, the drawn pixels are aligned correctly

sn4k3, 2024-12-20:
Yes, I know about that, is not easy to make it right, they are two different draw engines and even with offset to make it on correct place it will displacement in some cases

sn4k3, 2026-09-15:
Will be fixed on next release

closed 2026-09-15 by sn4k3