sn4k3/UVtools #967: [Crash] System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (org.freedesktop.DBus.Error.ServiceUnknown: The name com.canonical.AppMenu.Registrar was not provided by any .service files)

TheMiNuS, 2024-12-21:
### System

```Text
UVtools v5.0.2 X64
Operative system: Debian GNU/Linux 12 (bookworm) X64
Processor: Intel(R) Core(TM) i7-10700K CPU @ 3.80GHz
Processor cores: 16
Memory RAM: 17.82 / 62.57 GB
Runtime: linux-x64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.3
OpenCV: 4.9.0

Path:       /tmp/.mount_UVtoolB6d8iu/usr/bin/
Executable: /tmp/.mount_UVtoolB6d8iu/usr/bin/UVtools
```

### Printer and Slicer

- Printer:  Elegoo Saturn 2
- Slicer: Chitubox

### Description of the bug

```
System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (org.freedesktop.DBus.Error.ServiceUnknown: The name com.canonical.AppMenu.Registrar was not provided by any .service files)
 ---> Tmds.DBus.Protocol.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name com.canonical.AppMenu.Registrar was not provided by any .service files
   at Tmds.DBus.Protocol.DBusConnection.MyValueTaskSource`1.System.Threading.Tasks.Sources.IValueTaskSource.GetResult(Int16 token)
   at Tmds.DBus.Protocol.DBusConnection.CallMethodAsync(MessageBuffer message)
   at Tmds.DBus.Protocol.Connection.CallMethodAsync(MessageBuffer message)
   --- End of inner exception stack trace ---
Category: Task
File: UVtools_demo_file.sl1s  [Version: 0] [Class: SL1File]

Machine date time: 12/21/2024 3:20:43 PM
    UTC date time: 12/21/2024 2:20:43 PM
```

### How to reproduce

Open two gerber files (main PCB "Front or Back" and Holes "PTH") and select invert on the main pcb.
Select create one layer per gerber file.
It will either freeze or crash.

### Files

[GerberFiles.zip](https://github.com/user-attachments/files/18218779/GerberFiles.zip)

TheMiNuS, 2024-12-21:
Additional comments, the gerber files are generated with Kicad 8 (version Debian Backport). I run UVTools as an App-image on a Debian 12 Vanilla.

I have instabilities when I put multiple gerber files and start to invert colors individualy.

I can make it work but I need to generate an individual chitubox file and manage colors individually in each of them. It works but it could be nice to have both drills and PCB merged together.

sn4k3, 2024-12-21:
Duplicate of: https://github.com/sn4k3/UVtools/issues/964

TheMiNuS, 2024-12-21:
So let's close it but if you can also take a look at the color inversion bug when you have multiple gerber files in a PCB exposure with color inversion per gerber file.
I'm not sure it's the same bug as I do have it since a long time. It's not related to this version.

closed 2024-12-21 by TheMiNuS

sn4k3, 2024-12-21:
> look at the color inversion bug when you have multiple gerber files

You mean the instability or a not expected behavior of the inversion process?

TheMiNuS, 2024-12-21:
The app freeze.
Open two gerber files from my upload (one front or back and the NTH) and select invert for the pcb front or back.
It will freeze the app.

sn4k3, 2024-12-21:
Thats related to the bug you open and because UI framework. It affects only linux users, mostly debian based.

TheMiNuS, 2024-12-21:
Ok thx for your support. I hope we will have a fix soon.
I close that one as it's a duplicated bug declaration.