sn4k3/UVtools #974: [Bug] UV Tools corrupted .goo files.

metsamyyra, 2025-01-04:
### System

```Text
UVtools v5.0.3 X64
Operative system: Microsoft Windows 10.0.19045 X64
Graphic card: 
Processor: Intel(R) Core(TM) i5-4690K CPU @ 3.50GHz
Processor cores: 4
Memory RAM: 6.70 / 15.84 GB
Runtime: win-x64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.3
OpenCV: 4.9.0

Path:       C:\Program Files\UVTools\
Executable: C:\Program Files\UVTools\UVtools.exe
```

### Printer and Slicer

- Slicer: Chitubox

### Description of the bug

supported models in Chitubox
sliced
loaded into UVTools 
repaired issues using autorepair and manual drawing , and set printing parameters like always (worked already 4 times)
saved into -goo like always
printer doesn't recognize the .goo file in stick
try to reopen the same file (saved before into hard drive)

"Error opening the file"
"(1) RLE for layer 10 is corrupted, should start with 85 but got 65
 (2) Decoded RLE for layer 0 is corrupted, expected checksum <218>, got <150>
 (3) RLE for layer 20 is corrupted, should start with 85 but got 212
  (OK) " 
and thus does nothing.

IS my file lost? can it be repaired somehow?
Chitu won't open it again in -goo format even in slice form.

### How to reproduce

don't know , this has not happened before

sn4k3, 2025-01-04:
After a corruption file is lost.
Was the file open and saved while it was in the USB?

metsamyyra, 2025-01-05:
Thanks for the answer, I may have found the reason for the corruption - the file was saved first in the HD, but it ran out of space and finished saving automatically after freeing space. Still UVTools was open and everything was working, but I think it might have altered the parameters as I saved the same file then into the USB. Didn't just copy+paste it there though, used save as. 
Redid everything with the same parameters and it worked without a problem.

sn4k3, 2025-01-05:
Make sure to never run out of space in your HDD, it can corrupt disk as well. 
If SSD is even worse as low space degrade your SSD quick and make it slow.

closed 2025-01-07 by sn4k3