sn4k3/UVtools #980: [Bug] Gerber excellon drill file problem

ipeedie, 2025-01-16:
### System

```Text
UVtools v5.0.5 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Core(TM) i7-7820HQ CPU @ 2.90GHz
Processor cores: 8
Memory RAM: 9.93 / 31.85 GB
Runtime: win-x64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.3
OpenCV: 4.9.0

Sreens, resolution, working area, usable area:
1: 3840 x 2160 @ 150% (On this)
    WA: 3840 x 2160    UA: 2560 x 1440
2: 3840 x 2160 @ 150% (Primary)
    WA: 3840 x 2160    UA: 2560 x 1440

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: PCB.goo [Version: 0] [Class: GooFile]
```

### Printer and Slicer

### Description of the bug

I'm using Altium Circuit Studio 1.5.2 which generates drill files as per DrillExample.TXT below.

[DrillExample.TXT](https://github.com/user-attachments/files/18437233/DrillExample.TXT)

UVTools will not open the drill file unless I rename with *.DRL extension and remove the spindle speed and feed rate from the tool definitions as per DrillExampleFixed.DRL below (changed from *.DRL to *.TXT to allow attachement)...

[DrillExampleFixed.TXT](https://github.com/user-attachments/files/18437284/DrillExampleFixed.TXT)

Could UVTools handle this without manual intervention?

### How to reproduce

Drag and drop sliced file for printer into UVTools and then drag and drop drill file.

ipeedie, 2025-01-16:
I just found a random drill file from Altium and it appears to have the same format (and problem) as Circuit Studios fixed by removing spindle speed and feed rate from tool definitions.

ipeedie, 2025-01-16:
There is more to this than I thought. I have attached DRILL.DRL and LAYER.GBL which all look good and line up in Viewmate but when I try and generate a PCB exposure via UVTools I have problems. E.g.

1. If try and combine them (to subtract drill from layer) then UVTools hangs up indefinitely.
2. If try and generate only DRILL.DRL then I get attached PCB.goo. Comparing this to DRILL.DRL in viewmate there are several holes in the wrong place. E.g. several holes have appeared near the top of PCB.goo (moved from elsewhere).

I have not found a step by step instructions for subtracting a drill file from a copper layer. Am I doing something obviously wrong?

sn4k3, 2025-01-17:
For drill file to work, you must check "Draw all gerber files into same layer". 
Just add the two files, check that and drills will me made.

I'm not experiencing your 1. and 2. points. 
But your PCB.goo looks like to have bugs

ipeedie, 2025-01-17:
Thanks Tiago...

FYI, I am starting every time with a fresh ROOK.goo sliced file for Mars 4 Ultra from https://www.elegoo.com/pages/download.

For the spindle/feed problem if I drag and drop DrillExample.TXT (changed to *.DRL) onto UVTools, make no changes in the PCB exposure window, generate PCB exposure, generate PCB traces then I get the error below.

[image]

DrillExample.TXT is the drill file directly from Altium/Circuit Studio. I have to modify it as per DrillExampleFixed.TXT before UVTools (appears) to accept it.

ipeedie, 2025-01-17:
For the hole errors I can reproduce by loading ROOK.goo, dragging only DrillExampleFixed.TXT into UVTools, selecting Invert checkbox and Mirror then generating PCB exposure. The image below shows the drill file open in ViewMate (gerber viewer) at top and UVTools at bottom. I have highlighted in red a couple of the differences. I think I can see these errors in the image you provided too?

[image]

ipeedie, 2025-01-17:
I thought that problems might be arising from DrillExampleFixed having holes relative to origin outside of the 153.36x77.76mm available on the Mars 4 Ultra so I moved my origin as per Drill2.TXT below however this file shows similar issues.

[Drill2.TXT](https://github.com/user-attachments/files/18451555/Drill2.TXT)

sn4k3, 2025-01-17:
> For the spindle/feed problem if I drag and drop DrillExample.TXT (changed to *.DRL) onto UVTools, make no changes in the PCB exposure window, generate PCB exposure, generate PCB traces then I get the error below.
> 
> [image]
> 
> DrillExample.TXT is the drill file directly from Altium/Circuit Studio. I have to modify it as per DrillExampleFixed.TXT before UVTools (appears) to accept it. The modification is from...

Ok, I found it, it will be fixed on next release.

> For the hole errors I can reproduce by loading ROOK.goo, dragging only DrillExampleFixed.TXT into UVTools

I see it too, the file is using a non fixed format for decimals causing misplacement, I will try to fix that. Meanwhile if you can set to export as real units (With decimal point `.`)

ipeedie, 2025-01-17:
Great thanks Tiago.

The only drill file options Circuit Studios gives me are shown below, it doesn't look like it can give a decimal point just the standard  4:2, 4:3 and 4:4 formats. I'll just continue to etch PCBs without pilot holes for the time being.

Many thanks for you time.

[image]

ipeedie, 2025-01-17:
FYI I have attached 3x files for the 3x Leading/Trailing Zeroes options. UVTools seems to behave differently depending on which is used...

[KeepLeadingAndTrailingZeroes.TXT](https://github.com/user-attachments/files/18459138/KeepLeadingAndTrailingZeroes.TXT)

[SuppressLeadingZeroes.TXT](https://github.com/user-attachments/files/18459139/SuppressLeadingZeroes.TXT)

[SuppressTrailingZeroes.TXT](https://github.com/user-attachments/files/18459141/SuppressTrailingZeroes.TXT)

ipeedie, 2025-01-17:
I'm close.

I manually edited my drill file "Fixed.DRL" below to include a decimal point. I've also included the matching "Gerber.GBL".

[Fixed.TXT](https://github.com/user-attachments/files/18459553/Fixed.TXT)

[Gerber.TXT](https://github.com/user-attachments/files/18459556/Gerber.TXT)

My procedure for generating the exposure is below...

1. Open a valid Mars 4 Ultra sliced *.goo file in UVTools.
2. Drag both DRL and gerber together into UVTools.
3. Invert the gerber layer.
4. Toggle to Draw all gerber files into same layer.
5. Select Invert Colour (tracks should be black).
6. Generate

This produces UNMIRRORED.goo below which looks correct but this is for the bottom side of a PCB so needs mirrored.

[UNMIRRORED.TXT](https://github.com/user-attachments/files/18459572/UNMIRRORED.TXT)

If I use the procedure above but select the Mirror option then things look correct during preview but after generation I get MIRRORED.goo below where it looks like the gerber has mirrored but the drill file has not?

[MIRRORED.TXT](https://github.com/user-attachments/files/18459597/MIRRORED.TXT)

I could obviously mirror things in my PCB CAD as a workaround but it would simplify my workflow if UVTools handled it?

sn4k3, 2025-01-18:
If UVtools can't mirror properly you can always generate normal and use Tool - Mirror

sn4k3, 2025-01-31:
Should be fixed, please test.

ipeedie, 2025-02-01:
Works great, thanks, donation incoming...

closed 2025-02-01 by sn4k3