sn4k3/UVtools #983: [Bug]

nathanmahnke, 2025-01-17:
### System

```Text
UVtools v5.0.5 X64
Operative system: Microsoft Windows 10.0.26100 X64
Processor: AMD Ryzen 7 9800X3D 8-Core Processor           
Processor cores: 16
Memory RAM: 12.93 / 31.11 GB
Runtime: win-x64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.3
OpenCV: 4.9.0

Sreens, resolution, working area, usable area:
1: 3440 x 1440 @ 125% (Primary) (On this)
    WA: 3440 x 1380    UA: 2752 x 1104

Path:       C:\Program Files (x86)\UVtools_win-x64_v5.0.5\
Executable: C:\Program Files (x86)\UVtools_win-x64_v5.0.5\UVtools.exe
Loaded file: dwarf-1-presupported.pwsz [Version: 0] [Class: AnycubicZipFile]
```

### Printer and Slicer

- Printer: Anycubic Photon Mono M7 Pro
- Slicer: AnycubicPhotonWorkshop

### Description of the bug

Once a file is saved after issues are repaired, the volume of resin is multiplied by 1000.

### How to reproduce

[image]
After clicking save or save as, the volume property is overwritten during the save process. Before saving, the volume updates as expected when removing islands and such, but once the file is saved, the error occurs. The printer throws an error about the amount of resin being used when attempting to print with the generated file.

sn4k3, 2025-01-18:
Attach the file before save with UVtools.

kolmgorov, 2025-03-22:
I can confirm, this is also happening with UV tools when I save a file for printing on my M7 pro. The Resin volume went from 85.874ml, to 8,5874ml

wfcook, 2025-05-23:
I can confirm also. The resin volume for one file went from 114.50492 to 98983.03

Brand new M7 thinks it has doine one print and cranked out nearly 100L worth of plastic.

sn4k3, 2025-05-23:
Please provide me with a file, a original without the problem and the one modified with UVtools

LionKing2205, 2025-05-29:
I have same problem (((

LionKing2205, 2025-05-29:
> Please provide me with a file, a original without the problem and the one modified with UVtools

[exs.zip](https://github.com/user-attachments/files/20503922/exs.zip)

closed 2025-05-30 by sn4k3