sn4k3/UVtools #984: [Bug] Transition Layer Exposure Time Discrepancy between UVtools and Reality

Chainsoroth, 2025-01-18:
### System

```Text
UVtools v5.0.5 X64
Operative system: Microsoft Windows 10.0.22631 X64
Processor: AMD Ryzen 9 7950X3D 16-Core Processor          
Processor cores: 32
Memory RAM: 15.44 / 63.12 GB
Runtime: win-x64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.3
OpenCV: 4.9.0

Path:       C:\Users\dauer\UV Tools\
Executable: C:\Users\dauer\UV Tools\UVtools.exe
Loaded file: TransTest.ctb [Version: 5] [Class: CTBEncryptedFile]
```

### Printer and Slicer

- Printer: Elegoo Saturn 4 Ultra (FW 41.28)
- Slicer: ChituBox (v2.3)

### Description of the bug

UVtools is showing wrong transition layer exposure times for a .ctb and .goo file sliced in ChituBox. 
Print Settings: 2 Bottom Layers @ 20s, 5 transition layers, 2,1sek normal layers
Expected behaviour: Two layers @ 20s and then 5 Layers decreased by 2,98s every Layer.

Times shown in UVtools:
Layer 0: 20s
Layer 1: 20s
Layer 2: 17,76s
Layer 3: 15,52s
Layer 4: 13,29s
Layer 5: 11,05s
Layer 6: 8,81s
Layer 7: 2,1s

[image]

Times measured with a stopwatch at the printer:
Layer 0: 20s
Layer 1: 20s
Layer 2: 17,3s
Layer 3: 14,1s
Layer 4: 10,9s
Layer 5: 8,1s
Layer 6: 5,1s
Layer 7: 2,1s

Regarding my reaction times in handling the stopwatch, the measured data seems to be completely in the range of expected values (2,98s decrease in exposure time per Layer). 

But UVTools shows completely different values. 

### How to reproduce

Shows up in every opened .ctb .goo file. (See attached file)

### Files

[TransTest.zip](https://github.com/user-attachments/files/18464100/TransTest.zip)

sn4k3, 2025-01-18:
It's expected to have a small difference. Your printer uses hardware transition layers, the time is controlled by firmware and not by file. As so UVtools recreates the exposure time based on a simple match which can or not match the printer math for transition exposure time.
We have no control over it, other than selecting the number of transition layers, printer will do the rest...

Chainsoroth, 2025-01-18:
In other words the math/formula/calculation in UVtools is faulty. 

And I think I have an idea, what is wrong about it.

I (and Chitubox it seems) would calculate the time decrease per transition layer as followed:
("Bottom Layer Time" - "Normal Layer Time")/("# of Transition Layers" + 1)  --> (20s-2,1s)/(5+1) = 2,98s

And it seems that UVtools calculates:
("Bottom Layer Time" - "Normal Layer Time")/(_**"# of bottom Layers"**_ + "# of transition layers" + 1) --> (20s-2,1s)/(2+5+1) = 2,24s

I testet it by increasing the number of bottom layers and the shown transition layer decrease time in UVtools also gets smaller.

sn4k3, 2025-01-18:
UVtools calculates the step like you have in chitu formula:

https://github.com/sn4k3/UVtools/blob/27f46ff365efb1c5448aaca69b96d54ce753d1bc/UVtools.Core/FileFormats/FileFormat.cs#L5254-L5259

If you go to layer 2 you can see the information is right both in UVtools and File

[image]

The bottom information has different information, which don't matter much as that is just a representation, but I will take a look why does it show differently. However you should get what layer table says.

Chainsoroth, 2025-01-19:
Thank you for the further clarification! 

To conclude the findings:
ChituBox saves "incorrect" transition layer exposure times into the layer data.
BUT, it is of no consequence due to my (most?) printers use firmware based calculations for the transition layer exposure times, which leads to correct exposure times(/correct decrease in exposure time).

UVtools displays the decrease in exposure times which it finds between the last bottom layer and the first transition layer. Thus showing "technically correct" data.

Again, thanks for clarifying and sorry for my persistence. My lack of knowledge really annoyed me and I needed to understand the connections between everything.

closed 2025-01-20 by sn4k3

sn4k3, 2025-01-20:
Your printer can use custom transition times set by you, but the default for most printers that offer native transition is hardware.
Using your own time will require set a flag but I do not recommend just for it.
The transition time is just to smooth the time diference reducing a brute exposure diference and avoid potential visual artifacts, it does not matter for prints on a raft. It's more useful for prints direct on build plate, and even then, as you use 2 bottom layers and low bottom exposure you will not get any benefit out of it.
If your measurements are correct, the difference is not even worthy of such concern.