sn4k3/UVtools #986: [Crash] System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (Your system memory RAM hit the limit

kalvaer, 2025-01-20:
### System

```Text
UVtools v5.0.5 X64
Operative system: Microsoft Windows 10.0.26100 X64
Processor: 12th Gen Intel(R) Core(TM) i7-12700KF
Processor cores: 20
Memory RAM: 27.35 / 31.75 GB
Runtime: win-x64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.3
OpenCV: 4.9.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
```

### Printer and Slicer

- Printer: Elegoo Mars 4 Ultra
- Slicer: ChituBox Basic V2.3

### Description of the bug

```
System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (Your system memory RAM hit the limit of 31.24GB from a total of 31.75GB.  Available: 0.51GB.  UVtools: 24.98GB (78.68%).
The program crashed on purpose due the impossibility to pause or cancel the running operation, this was to relief pressure and ensure the system stability.

Monitor your memory RAM under the system tools and re-run the operation to check the usages and find the cause.
You may need to stop other processes or increase your memory RAM in order to deal with huge files and/or heavy operations.

Currently you have configured a lower limit of 1GB of available memory RAM for the operations.)
 ---> System.InsufficientMemoryException: Your system memory RAM hit the limit of 31.24GB from a total of 31.75GB.  Available: 0.51GB.  UVtools: 24.98GB (78.68%).
The program crashed on purpose due the impossibility to pause or cancel the running operation, this was to relief pressure and ensure the system stability.

Monitor your memory RAM under the system tools and re-run the operation to check the usages and find the cause.
You may need to stop other processes or increase your memory RAM in order to deal with huge files and/or heavy operations.

Currently you have configured a lower limit of 1GB of available memory RAM for the operations.
   at UVtools.UI.MainWindow.<>c__DisplayClass57_0.<RamUsageTimerOnElapsed>b__0() in UVtools.UI\MainWindow.axaml.cs:line 518
   at Avalonia.Threading.DispatcherOperation.InvokeCore()
   --- End of inner exception stack trace ---
Category: Task
File: Basic.stl_0.05_2.5_2025_01_20_00_56_00.ctb  [Version: 5] [Class: CTBEncryptedFile]

Machine date time: 20 Jan 2025 01:11:48
    UTC date time: 20 Jan 2025 00:11:48
```

### How to reproduce

1. Open file
2. Click Calibration -> Stress Tower
3. Default Settings.
4. Click Stress Tower on the bottom
5. Click Yes to generate tower

(Tried using files from different slicers, but the same thing happens)

### Files

[image]
[image]

sn4k3, 2025-01-20:
Stress tower uses very memory.
You can disable the protection on UVtools settings.
Read error message...

sn4k3, 2025-02-01:
Should be fixed

closed 2025-02-01 by sn4k3