sn4k3/UVtools #990: [Bug] extreme mousewheel sensitivity when zooming

thirstyone, 2025-01-21:
### System

```Text
UVtools v5.0.5 X64
Operative system: Linux Mint 22 X64
Processor: Intel(R) Core(TM) i7-2675QM CPU @ 2.20GHz
Processor cores: 8
Memory RAM: 15.11 / 15.53 GB
Runtime: linux-x64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.3
OpenCV: 4.9.0

Sreens, resolution, working area, usable area:
1: 1440 x 900 @ 100% (On this)
    WA: 1440 x 867    UA: 1440 x 867

Path:       /tmp/.mount_UVtooluVFCpK/usr/bin/
Executable: /tmp/.mount_UVtooluVFCpK/usr/bin/UVtools
Loaded file: espdashtry2 (5.0.5, inverted).svgx [Version: 0] [Class: FlashForgeSVGXFile]
```

### Printer and Slicer

- Printer: 
- Slicer: 

### Description of the bug

I'm using UVtools for pcb exposure. It's way too difficult to zoom in to a needed level with the mousewheel - it's either too big or too small, and it takes quite an effort to get into the right zoom level. Is there a way to make the mousewheel less sensitive? 

### How to reproduce

open any pcb, try to zoom in/out

sn4k3, 2025-01-21:
Each zoom tick will increase zoom by 0.25x / 0.5x / 1x / 4x depending on your current zoom level.
Zoom sensibility comes from mouse (If you use a free scroll well style) or from system. Program just catch each tick event... A tick is when you fell the mouse wheel to "click" in each movement.

For me under linux I can easy use zoom with precision to whatever level.
This is either your mouse or system settings

thirstyone, 2025-01-22:
Indeed, I tried it on another linux mint system, with a real mouse, and it worked just fine. I connected that mouse to my initial system - and it also worked fine. So, the problem is narrowed down to my touchpad (I use the same two-finger 'scroll' gesture to zoom in and out, that I use to scroll content in browsers). However, there are couple of points: 
1. this gesture works just fine with other software (e.g. kicad, system interface) - the problem is just with UVtools. 
2. i couldn't find anything related to trackpad in this regard in system settings. 

so, may be, it has something to do with the underlying framework?

sn4k3, 2025-01-29:
Today I tried on my laptop, and it work good with trackpad, on Windows. I also tested on Mint under a VM and work as same as the windows version.
I guess this is related to the sensitivity of the gesture

mnemnion, 2025-02-09:
+1, trying to scroll using the stock touchpad on macOS is unusable, the slightest gesture scrolls by many levels and I have been unable to get it to scroll incrementally no matter how delicately I try.

Adjusting the system-level sensitivity is not an acceptable fix here, the touchpad works correctly with every other program I use so I won't sacrifice that to make UVTools usable.  

What would plausibly work is making the number of 'ticks' per zoom level itself configurable.   The problem might also be with the framework, or how it's configured for macOS builds, none of that is clear to me.

I don't intend to come across as demanding or ungrateful, it's just a bug after all.  I want to make it clear that the current behavior makes scrolling out to get context on islands _impossible_ using my touchpad, and that affects my ability to use UVTools to detect and support islands.  I don't have any modifications to how the operating system interprets the scroll gesture, so it seems plausible that anyone using a Mac touchpad is going to find things similarly unusable.  

I wasn't able to find keystrokes to effect the change in zoom level, or UI controls for it other than scrolling itself.  If there's a way to do it by pressing keys, I would prefer that anyway, so if anyone knows how to do that, it would be a useful thing to add to this issue while it remains open.

sn4k3, 2025-02-12:
Here the 30ms version: [UVtools.AvaloniaControls_30ms.zip](https://github.com/user-attachments/files/18760695/UVtools.AvaloniaControls_30ms.zip)

sn4k3, 2025-02-12:
You can keep adding time without negative effect up to the image redraw time. Each zoom needs to redraw the image region and that take a little time based on image resolution. However, with 50ms I already notice a non-fluid scroll on my Windows if I spam the scroll wheel, which does not feel natural to me. The truth is that UVtools was not made with laptop's in mind.

[UVtools.AvaloniaControls_40ms.zip](https://github.com/user-attachments/files/18762090/UVtools.AvaloniaControls_40ms.zip)
[UVtools.AvaloniaControls_50ms.zip](https://github.com/user-attachments/files/18762091/UVtools.AvaloniaControls_50ms.zip)

sn4k3, 2025-02-13:
Try this file, I've disable normal event and subscribe only to a touchpad gesture. Also I removed the debounce to see how it works.
Additionally it will create a file on your desktop, do some zoom in/out and report the findings plus the txt file contens

[UVtools.AvaloniaControls_touchpadonly.zip](https://github.com/user-attachments/files/18776180/UVtools.AvaloniaControls_touchpadonly.zip)

thirstyone, 2025-02-13:
Hey, thank you for your effort! 
The last one didn't work at all. I tried the 40ms version, and it seems to have solved my problem. 
Now, I wonder if I have to stick to my current version or to manually update the file with each new version of the app.

sn4k3, 2025-02-13:
Ok, so TouchpadGesture not working here.
BTW can you try the gesture to move horizontal scroll bar on touchpad? My guess is that wont work.

Also your txt give me the final clue. The reason why current implementation is zooming crazy is because your increments are very tiny, you have deltas of 0.02 and 0.01, and the code ignore delta value because it uses Ticks instead of real zoom. So each time you event fires it increate a tick which in mouse wheel deltas are 1.0.
I will try a new patch to see the result...

mnemnion, 2025-02-13:
It's difficult to check horizontal scroll in the region of interest, because it starts zoomed out, and with the zoom keys also broken I can't zoom in to try it.  Both horizontal scroll and vertical scroll work correctly inside the properties and groups field on the sidebar, but if I recall correctly, that was always true.

Here's the log from that experiment, in case it's useful.

[UVtools_Scroll.txt](https://github.com/user-attachments/files/18788619/UVtools_Scroll.txt)

sn4k3, 2025-02-13:
Test this, I implemented horizontal scroll with compatible mouses and this implements native zoom using deltas. Log file also removed. 

[UVtools.AvaloniaControls_native.zip](https://github.com/user-attachments/files/18788669/UVtools.AvaloniaControls_native.zip)

sn4k3, 2025-02-13:
What about horizontal scroll gesture?

sn4k3, 2025-02-13:
Here another version. It zoom native, but if you press ALT it will zoom with levels. Both are using debounce to 20ms (This will be configured into UI settings).
If you hold CTRL, it will do vertical scroll instead.
Report back

[UVtools.AvaloniaControls_rc.zip](https://github.com/user-attachments/files/18789921/UVtools.AvaloniaControls_rc.zip)

mnemnion, 2025-02-13:
When I said it was native I meant to also convey that horizontal scrolling worked.  

With the latest version, horizontal two-finger gestures move the screen horizontally, vertically moves in and out.  With the option / Alt key held down, it becomes very fast. 20ms is pretty twitchy on my machine, but as you said, configurable.

Control + scroll does something else entirely, it zooms the entire screen in and out.  That happens for any application though, so I think the OS is intercepting what you intended it to do.   I don't believe that's a custom configuration on my part, or at least, I wasn't able to find it in Settings if so.

sn4k3, 2025-02-13:
Ok so I think this will be the final. Will be configurable, so people can fine tune...

sn4k3, 2025-02-15:
Test new release and report back.
Need to go Settings -> Layer Preview

closed 2025-03-10 by sn4k3