sn4k3/UVtools #997: [Bug] Printing stopped and satur 4 freezing when adding new parameters with uvtools

remikrol, 2025-02-07:
### System

```Text
UVtools v5.0.6 X64
Operative system: Microsoft Windows 10.0.22631 X64
Processor: AMD Ryzen 7 5700U with Radeon Graphics         
Processor cores: 16
Memory RAM: 10.18 / 13.86 GB
Runtime: win-x64
Framework: .NET 9.0.1
AvaloniaUI: 11.2.3
OpenCV: 4.10.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 125% (Primary) (On this)
    WA: 1920 x 1080    UA: 1536 x 864
2: 1920 x 1080 @ 100%
    WA: 1920 x 1080    UA: 1920 x 1080

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: CTUVTOOLS.goo [Version: 0] [Class: GooFile]
```

### Printer and Slicer

- Printer: Elegoo saturn 4
- Slicer: lychee slicer

### Description of the bug

Hi the communauty, 

I sliced my STL file using Lychee and then opened it in UVTools to modify the "wait time before cure" for the bottom layer and some normal layers.

When I tried printing on my Saturn 4, the machine froze just before the UV exposure of the first layer.

I have successfully used UVTools, LycheeSlicer, and the Saturn 4 for other prints before, but I had never changed print parameters in UVTools before printing.

Does anyone know how to fix this issue?

Thanks

### How to reproduce

Open you sliced file.
Go to the shield icon
Go to configure suggestions
GO to "wait time before cure"
Change parameters and save
Re-open the file
Go to the shield
Apply the new "wait time before cure"
Save the new file
Print-it

sn4k3, 2025-02-07:
That modification turns a flag on which is known to cause trouble on Anycubic machines, but never heard of trouble for goo files.
Can you try to convert to CTB and CTB Encrypted to see which one works?

remikrol, 2025-02-07:
I can convert with uvtools, but unfortunately my printer doen't recognize CTB files - only Goo

sn4k3, 2025-02-07:
On the bug file, reopen with UVtools, go to File - Reset layer properties. Save and try to print. Does it work now?

remikrol, 2025-02-10:
Hello,

When I perform "Reset Layer Properties" on a UVTools Goo saved file, a confirmation box appears. I click "Yes," but no changes seem to be applied to the layer (according to the layer data window).

For reference, I successfully printed another file with UVTools without modifying the print parameters, and the print job completed correctly.

If I take the original file from Lychee and apply the desired parameters, the layer data displays the following values:

First Layer:
Index: 0 (Modified)
LayerHeight: 0.050 mm
PositionZ: 0.050 mm
IsBottomLayer: True
ExposureTime: 0.01 s
LiftHeight: 4 mm @ 5 mm/min
LiftHeight2: 4 mm @ 20 mm/min
RetractHeight: 4 mm @ 70 mm/min
According to the view window, this first layer appears empty.

Second Layer:
Index: 1 (Modified)
LayerHeight: 0.050 mm
PositionZ: 0.050 mm
IsBottomLayer: True
ExposureTime: 0.01 s
LiftHeight: 0.1 mm @ 5 mm/min
LiftHeight2: 0 mm @ 20 mm/min
RetractHeight: 0 mm @ 70 mm/min
In the view window, the first layer still appears empty, but the second layer looks fine.

Expected Layer Information:
LayerHeight: 0.050 mm
PositionZ: 0.050 mm
IsBottomLayer: True
ExposureTime: 40.00 s
LiftHeight: 4 mm @ 5 mm/min
LiftHeight2: 4 mm @ 20 mm/min
RetractHeight: 4 mm @ 70 mm/min
Both the first and second layers should be visible (not empty).

For reference, before modifying the parameters, the exposure time was 30 s. All layers were visible.

Hope this information is helpful to you

remikrol, 2025-02-10:
I ran a new test:

I sliced a file with Lychee, added it to UVtools without making any changes, and saved it.

After printing the first layer, the printer froze.

I had previously printed this part with Lychee without any issues, and the printer didn’t freeze.

sn4k3, 2025-02-10:
> When I perform "Reset Layer Properties" on a UVTools Goo saved file, a confirmation box appears. I click "Yes," but no changes seem to be applied to the layer (according to the layer data window).

But did you try to save the file and try to print it?

sn4k3, 2025-02-10:
> I ran a new test:
> 
> I sliced a file with Lychee, added it to UVtools without making any changes, and saved it.
> 
> After printing the first layer, the printer froze.
> 
> I had previously printed this part with Lychee without any issues, and the printer didn’t freeze.

Send me that both files, the original and the uvtools saved

remikrol, 2025-02-10:
[CT4UVTOOLS.zip](https://github.com/user-attachments/files/18738371/CT4UVTOOLS.zip)

Sorry, i only can send you on zip format. 

 2 files in the package : CT4 (from lychee) et CT4uvtools (whit modification made by uvtools)

sn4k3, 2025-02-10:
Have you attached the correct file? I have compared both and there are no visible differences:

```
D:\Tiago\Desktop> UVtoolsCmd compare .\CT4.goo .\CT4UVTOOLS.goo
Opening file CT4.goo: Done in 2.23s
Opening file CT4UVTOOLS.goo: Done in 0.01s
[Global #7]
MaterialGrams: 54022.59 != 53.99
TemporaryOutputFileFullPath: D:\Tiago\Desktop\CT4.goo.tmp != D:\Tiago\Desktop\CT4UVTOOLS.goo.tmp
FileFullPath: D:\Tiago\Desktop\CT4.goo != D:\Tiago\Desktop\CT4UVTOOLS.goo
FileFullPathNoExt: D:\Tiago\Desktop\CT4 != D:\Tiago\Desktop\CT4UVTOOLS
Filename: CT4.goo != CT4UVTOOLS.goo
FilenameNoExt: CT4 != CT4UVTOOLS
FilenameStripExtensions: CT4 != CT4UVTOOLS
```

Comparing files with binary show same difference as UVtools (Only Material grams)

[image]

Everything match with exception of MaterialGrams, which your original Goo have a large value, this variable alone can't be the cause of the issue.

remikrol, 2025-02-10:
[UVTOOLS ISSUE.zip](https://github.com/user-attachments/files/18739010/UVTOOLS.ISSUE.zip)<

sorry, maybe i share the wrong file. There is the new one. 

the main & only dfference concern the wait time before exposure. For the first file from lyche, i have this parameter : 

- 40s for bottom layer
- 5 s for normal layer

For the uvtools file : 

- 40 s for the layers betwenn 0&2.5mm
- 5s  for 2.5mm and final layer

sn4k3, 2025-02-10:
Try this in a dry print.

[CT4UVTOOLS_modded.zip](https://github.com/user-attachments/files/18739190/CT4UVTOOLS_modded.zip)

remikrol, 2025-03-13:
Hey @sn4k3, sorry for the late reply. I ran the test today, though not entirely—just the first 10% (15 minutes). Everything went well—no freezing, no issues.

At first, I thought it might be a hardware issue with my Saturn, but a few days ago, I printed another part and noticed freezing when using the goo file from UVTOOLS, but not with Lychee Slicer (same file, same settings).

sn4k3, 2025-03-13:
If my modded file works it mean you can't use the per layer settings or you will get trouble.
It turn the flag on if you change any layer value to be different from default.

To revert everything you need to go File -> Reset layer properties.
A issue also present on almost all anycubic machines

remikrol, 2025-03-13:
Ok,

You mean i can't change layer properties like wait time before cure or after cure with UVTOOLS?

sn4k3, 2025-03-14:
You can, if you apply global. However if you apply just to a layer group or individual layer or use the auto UVtools suggestion then it can trigger the problem.

sn4k3, 2025-07-28:
Fixed by disabling per layer settings by default on Goo

closed 2025-07-28 by sn4k3