sn4k3/UVtools #999: [Bug]

DrKropp, 2025-02-07:
### System

```Text
UVtools v5.0.6 X64
Operative system: Ubuntu 24.10 X64
Processor: AMD Ryzen 5 4500U with Radeon Graphics
Processor cores: 6
Memory RAM: 17.25 / 30.27 GB
Runtime: linux-x64
Framework: .NET 9.0.1
AvaloniaUI: 11.2.3
OpenCV: 4.10.0

Sreens, resolution, working area, usable area:
1: 3840 x 2160 @ 125% (Primary) (On this)
    WA: 3840 x 2112    UA: 3072 x 1690
2: 1440 x 2560 @ 125%
    WA: 1440 x 2112    UA: 1152 x 1690

Path:       /tmp/.mount_UVtool3tFKEl/usr/bin/
Executable: /tmp/.mount_UVtool3tFKEl/usr/bin/UVtools
Loaded file: UVtools_demo_file.sl1s [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: Anycubic Photon Mono X
- Slicer: Prusaslicer 2.9.0

### Description of the bug

When I attempt to open the SL1 file to convert it to PWMX I get the following error message:

System.IO.InvalidDataException: End of Central Directory record could not be found.
   at System.IO.Compression.ZipArchive.ReadEndOfCentralDirectory()
   at System.IO.Compression.ZipArchive..ctor(Stream stream, ZipArchiveMode mode, Boolean leaveOpen, Encoding entryNameEncoding)
   at System.IO.Compression.ZipFile.Open(String archiveFileName, ZipArchiveMode mode, Encoding entryNameEncoding)
   at UVtools.Core.FileFormats.SL1File.DecodeInternally(OperationProgress progress) in UVtools.Core/FileFormats/SL1File.cs:line 659
   at UVtools.Core.FileFormats.FileFormat.Decode(String fileFullPath, FileDecodeType fileDecodeType, OperationProgress progress) in UVtools.Core/FileFormats/FileFormat.cs:line 4616
   at UVtools.UI.MainWindow.<>c__DisplayClass99_0.<ProcessFile>b__2() in UVtools.UI/MainWindow.axaml.cs:line 1497

### How to reproduce

Slice a file and save the output to a flash drive. Launch UVTools and File>Open the sliced file

sn4k3, 2025-02-07:
I tried to auto convert and manually convert the SL1 to PWMX and it work fine.
Provide the problematic fine so I can test

DrKropp, 2025-02-13:
Please mark this as closed. I had a corrupted printer profile in PrusaSlicer and no longer had any issues after removing it. 

Thanks for the quick response and the amazing work you have done on this software!

closed 2025-02-13 by DrKropp