sn4k3/UVtools #1000: [Bug] Unable to detect PrusaSlicer nor SuperSlicer on your system.

mike-truebody, 2025-02-08:
### System

```Text
UVtools v5.0.6 X64
Operative system: Ubuntu 24.04.1 LTS X64
Processor: AMD Ryzen 5 3550H with Radeon Vega Mobile Gfx
Processor cores: 8
Memory RAM: 8.17 / 15.31 GB
Runtime: linux-x64
Framework: .NET 9.0.1
AvaloniaUI: 11.2.3
OpenCV: 4.10.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1000    UA: 1920 x 1000
2: 1920 x 1080 @ 100%
    WA: 1920 x 1000    UA: 1920 x 1000

Path:       /tmp/.mount_UVtool6eDSkk/usr/bin/
Executable: /tmp/.mount_UVtool6eDSkk/usr/bin/UVtools
Loaded file: UVtools_demo_file.sl1s [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: 
- Slicer: 

### Description of the bug

Unable to detect PrusaSlicer nor SuperSlicer on your system, please ensure you have latest version installed.
Was looking on: /home/mike-tb/.PrusaSlicer and /home/mike-tb/.SuperSlicer

Click 'Yes' to open the PrusaSlicer webpage for download
Click 'No' to dismiss

Prusa has now switched to using Flathub as the only official distribution  "Important note for Linux users: As noted before, we are now using [Flathub](https://flathub.org/) as the only official distribution channel of PrusaSlicer on Linux. The reasons behind this decision and what it means for you are described at the end of [2.9.0-alpha1 release log](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1)."

Due to this UV Tools will no longer find and install profiles into Prusa slice because it is looking in the wrong place. 

### How to reproduce

Install Prusa Slicer 2.0.0 on Linux (Ubuntu) using the new Flathub installation method.
Follow the setup guide as per the UV Tools GitHub (https://github.com/sn4k3/UVtools/wiki/Setup-PrusaSlicer)
In UV Tools when you go to Menu -> Help -> Install profiles into PrusaSlicer you will get the following error:
"Unable to detect PrusaSlicer nor SuperSlicer on your system, please ensure you have latest version installed.
Was looking on: /home/mike-tb/.PrusaSlicer and /home/mike-tb/.SuperSlicer

Click 'Yes' to open the PrusaSlicer webpage for download
Click 'No' to dismiss"

I believe the new installation location is /var/lib/flatpak/app/com.prusa3d.PrusaSlicer

sn4k3, 2025-02-15:
Should be fixed, please test and report

mike-truebody, 2025-02-15:
It now works perfectly, thank you very much.

closed 2025-02-15 by mike-truebody