sn4k3/UVtools #1001: [FEATURE REQUEST] Change material in Photon Mono 2

rmortes, 2025-02-10:
### Is your feature request related to a problem?

Hi! I have a piece I'd like to print in two colors. I haven't found a good way to pause the printer, change the resin and the resume the printing, and your software is kinda my last hope.

### Describe the solution you'd like

There's two ways in which I see this feature being implemented:
1. You can add a tool to insert an empty layer with like a reeeeealy long exposure time, or something like that, that just lifts the plate, waits 10 minutes, and continues printing. This would probably be the most straightforward way to implement it, but I don't know if you can retract the plate mid print like that, and it also relies in timing correctly the material change which makes this operation supervised
2. The other way I could see this feature being implemented is splitting the print in two. Say I have a 50mm cube that I want to print 60% black, 40% white. The input file gets turned into two output files: a 30mm tall 50x50mm box and a 20mm tall 50x50mm box that happens to start at 30mm height. This way I can print the first color, calmly change the resin, maybe clean a bit of the printed cube (without removing it from the build plate, ofc), reattach the build plate and then print the second color like a "new model"

I don't know if it's technically possible, or why Anycubic didn't include a "pause at layer" command into their file size, but I know you guys are my last hope

sn4k3, 2025-02-10:
None of that is possible with your printer.
For that you must have a true GCode printer or convert your printer to klipper

Your 1. and 2. fails because even if file format accept some different parameters per layer, activation such flag will cause file to fail to print as you can find multiple issues related to that.

rmortes, 2025-02-10:
Wellp, I guess it makes sense why apparently no one had done this with my printer online. Is it even possible to "install" klipper on my machine?

sn4k3, 2025-02-10:
> Is it even possible to "install" klipper on my machine?

Only if you in will to DIY and have some background with custom builds.

https://github.com/sn4k3/klipper-msla
https://github.com/[OleksandrPanforov/saturn-klipper](https://github.com/OleksandrPanforov/saturn-klipper)

rmortes, 2025-02-11:
Ok... That's waaay over my skill level at the moment, but thank you SO much for taking your time answering me. Cheers!