sn4k3/UVtools #1002: [Bug] System.NullReferenceException: Object reference not set to an instance of an object (Lychee 7.2 ctb v4)

crosswick, 2025-02-19:
### System

```Text
UVtools v5.0.7 X64
Operative system: Darwin 21.6.0 Darwin Kernel Version 21.6.0: Thu Jul  6 22:18:26 PDT 2023; root:xnu-8020.240.18.702.13~1/RELEASE_X86_64 X64
Processor: Intel(R) Xeon(R) CPU           X5680  @ 3.33GHz
Processor cores: 24
Memory RAM: 23.02 / 43.12 GB
Runtime: osx-x64
Framework: .NET 9.0.2
AvaloniaUI: 11.2.4
OpenCV: 4.10.0

Sreens, resolution, working area, usable area:
1: 3840 x 2160 @ 100% (Primary) (On this)
    WA: 3840 x 2135    UA: 3840 x 2135

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: elephant8_07m_10ml_copy.ctb [Version: 4] [Class: ChituboxFile]

General configuration for OpenCV 4.10.0 =====================================
  Version control:               3.4.2-10289-g3beff6e6c5

  Platform:
    Timestamp:                   2025-01-17T23:11:46Z
    Host:                        Darwin 21.6.0 x86_64
    CMake:                       3.31.4
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               Release

  CPU/HW features:
    Baseline:                    SSE SSE2 SSE3 SSSE3 SSE4_1
      requested:                 DETECT
    Dispatched code generation:  SSE4_2 FP16 AVX AVX2 AVX512_SKX
      requested:                 SSE4_1 SSE4_2 AVX FP16 AVX2 AVX512_SKX
      SSE4_2 (1 files):          + POPCNT SSE4_2
      FP16 (0 files):            + POPCNT SSE4_2 FP16 AVX
      AVX (3 files):             + POPCNT SSE4_2 AVX
      AVX2 (25 files):           + POPCNT SSE4_2 FP16 FMA3 AVX AVX2
      AVX512_SKX (2 files):      + POPCNT SSE4_2 FP16 FMA3 AVX AVX2 AVX_512F AVX512_COMMON AVX512_SKX

  C/C++:
    Built as dynamic libs?:      NO
    C++ standard:                11
    C++ Compiler:                /Library/Developer/CommandLineTools/usr/bin/c++  (ver 14.0.0.14000029)
    C++ flags (Release):         -fsigned-char -W -Wall -Wreturn-type -Wnon-virtual-dtor -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -Wno-deprecated-copy -O3 -DNDEBUG  -DNDEBUG
    C++ flags (Debug):           -fsigned-char -W -Wall -Wreturn-type -Wnon-virtual-dtor -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -Wno-deprecated-copy -g  -O0 -DDEBUG -D_DEBUG
    C Compiler:                  /Library/Developer/CommandLineTools/usr/bin/cc
    C flags (Release):           -fsigned-char -W -Wall -Wreturn-type -Wnon-virtual-dtor -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
    C flags (Debug):             -fsigned-char -W -Wall -Wreturn-type -Wnon-virtual-dtor -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
    Linker flags (Release):      -Wl,-dead_strip  
    Linker flags (Debug):        -Wl,-dead_strip  
    ccache:                      NO
    Precompiled headers:         NO
    Extra dependencies:          -framework OpenCL -framework Cocoa -framework AppKit
    3rdparty dependencies:       ittnotify libjpeg-turbo libpng libtiff libopenjp2 IlmImf zlib

  OpenCV modules:
    To be built:                 core highgui imgcodecs imgproc videoio
    Disabled:                    dnn features2d flann gapi ml photo video world
    Disabled by dependency:      calib3d objdetect stitching
    Unavailable:                 java python2 python3 ts
    Applications:                -
    Documentation:               NO
    Non-free algorithms:         NO

  GUI:                           COCOA
    Cocoa:                       YES
[... 154 lines cut ...]
099: System.Diagnostics.Process, Version=9.0.0.0
100: System.IO.Pipes, Version=9.0.0.0
```

### Printer and Slicer

- Printer: Elegoo Mars 2
- Slicer: Lychee 7.2

### Description of the bug

When I slice a ctb v4 file in Lychee, import it into UV Tools, generate an elephant foot calibration test and save it, the following error results:

`System.NullReferenceException: Object reference not set to an instance of an object.
   at UVtools.Core.FileFormats.ChituboxFile.EncodeInternally(OperationProgress progress) in UVtools.Core/FileFormats/ChituboxFile.cs:line 1845
   at UVtools.Core.FileFormats.FileFormat.Encode(String fileFullPath, OperationProgress progress) in UVtools.Core/FileFormats/FileFormat.cs:line 4559
   at UVtools.UI.MainWindow.<>c__DisplayClass105_0.<SaveFile>b__2() in UVtools.UI/MainWindow.axaml.cs:line 2279`

### How to reproduce

see above, and the file below. using ctb v3 does not produce the error. 

### Files

[Boxes_of_Calibration.ctb.zip](https://github.com/user-attachments/files/18868341/Boxes_of_Calibration.ctb.zip)

sn4k3, 2025-02-20:
I used your attached file and did the "generate an elephant foot calibration test and save it" but it does not raise any error to me. File was saved and reloaded fine. Are you sure you have the correct file attached?

crosswick, 2025-02-20:
Thank you for testing - I will doublecheck my file once I’m back home, tomorrow at the latest.

crosswick, 2025-02-20:
Fresh install on macOS Sequioa on different machine (iMac18,3): same issue

Reinstall on Mac Pro detailed above: same issue

Will try my third machine: macOS Sequoia, arm64

crosswick, 2025-02-20:
arm64 is a non-starter for me...

[image]

sn4k3, 2025-02-20:
The problem is with lychee slicer not setting a MachineName on file, resulting somehow on a `null` reading. However they define a machine name address... So machinename have an address but with 0 size. 
My bet is that if you try chitubox it will work

[image]

Windows and macOS have no difference in file format logic, there are no **IF** regarding OSes. 
Have you copied the problematic file or did you slice a new one under windows and lychee?

> macOS Sequoia arm64 is a non-starter for me...

Never manual install UVtools on macOS, always use the auto-installer script, stated on readme page.