sn4k3/UVtools #1004: [Bug] cannot open file, UVtools crashes when file -> open.

sooperwill, 2025-02-26:
### System

```Text
UVtools v5.0.7 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: AMD Ryzen 7 5800X 8-Core Processor             
Processor cores: 16
Memory RAM: 7.62 / 35.93 GB
Runtime: win-x64
Framework: .NET 9.0.2
AvaloniaUI: 11.2.4
OpenCV: 4.10.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: UVtools_demo_file.sl1s [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: saturn 4 ultra
- Slicer: chitubox

### Description of the bug

After installing uvtools with provided .msi, i try to open a file (file -> open) Open screen pops up, then immediately  "UVtools has stopped working" error. and is asked to close program.

### How to reproduce

File -> Open
open screens pops up, then error pops up.

sn4k3, 2025-02-26:
Have you tried to reinstall? This is not normal, also UVtools have a crash manager, if it does not show a error report from UVtools it probably something in your system.

sooperwill, 2025-02-26:
uninstalled it and reinstalled it. tried the zip as well. works fine on laptop. more than likely something with my system. will investigate

closed 2025-05-09 by sn4k3