sn4k3/UVtools #1009: [Bug] PCB exposure produces nothing

rhn, 2025-03-11:
### System

```Text
UVtools v5.0.5 X64
Operative system: Fedora Linux 40 (Forty) X64
Processor: Intel(R) Core(TM) i5-3380M CPU @ 2.90GHz
Processor cores: 2
Memory RAM: 11.15 / 11.44 GB
Runtime: linux-x64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.3
OpenCV: 4.9.0

Sreens, resolution, working area, usable area:
1: 1366 x 768 @ 100% (Primary) (On this)
    WA: 1300 x 736    UA: 1300 x 736

Path:       /tmp/.mount_UVtoolgODnEX/usr/bin/
Executable: /tmp/.mount_UVtoolgODnEX/usr/bin/UVtools
Loaded file: UVtools_demo_file.sl1s [Version: 0] [Class: SL1File]
```

### Printer and Slicer

Gerber file from KiCAD Version: 8.0.9-1.fc40, release build

### Description of the bug

After loading the Gerber file through the PCB plugin, the output has 0 layers.

The system info comes from 5.0.5, but 5.0.7 also does the same thing, and last 4.x version is also affected.

### How to reproduce

To reproduce:
1. Produce Gerber file using KiCAD Plot tool, output only F.CU layer
2. Open UVTools PCB Exposure
3. Select the .grb file (not .zip)
4. Confirm

Result: The model is replaced by black, layer slider doesn't work, attempts to save fail with some exception.

Expected: A single-layer image.

### Files

Attaching the latest gerber that failed.

[schematic_ic-F_Cu.zip](https://github.com/user-attachments/files/19195175/schematic_ic-F_Cu.zip)

sn4k3, 2025-03-11:
If your PCB shows on previewer but not after generate PCB it mean your origin is outside bounds of your printer screen size.
Adjust you PCB and align the top most left to be 0,0

rhn, 2025-03-11:
I'm not sure if this is what's happening. I seem to have garbage in the preview as well.
I adjusted the layout to be near the top left of the sheet in KiCAD and that didn't change anything.

sn4k3, 2025-03-11:
See how far your PCB is from 0,0:

[image]

rhn, 2025-03-11:
Oh wait, I loaded the same file again.

Moving to the corner actually makes this work.

I think the current behaviour is confusing, I wasted an hour before I hacked a workflow around Lithophane and Inkscape. Could it be improved?

When I send the Gerber file to a manufacturing service, they build a board relative to Edge.Cuts.

Perhaps place the printer image relative to Edge.Cuts rather than the drawing sheet?
Or if loading a lone CU .gbr file, then center the plot on the printing area?

sn4k3, 2025-03-11:
There are multiple reasons why your origin is respected here:

- Jigs (Placement on screen)
- Overlays
- Multi layers
- Holes

In the future I can make a option to auto align but will require the board outline file

closed 2025-03-11 by sn4k3

rhn, 2025-03-11:
I think some minimal error reporting could be a big improvement. The reason I thought UVTools was broken is that there was a nasty stack trace when saving he file. If an empty preview and an empty layer was generated I would be more likely to think that the file is the problem than the tool.
Even better if I got a dialog telling me the same as you just did in this issue: that the layer is empty maybe because the position is out of bounds.
UVTools knows the size of the printer area, so the tool should be able to detect that the loaded file contains stuff that doesn't fit.

Thanks for helping me out!

sn4k3, 2025-03-11:
I can include that message if any draw was made outside the screen and prevent the tool to run empty.
Also consider using a printer with square pixels for PCB usage as rectangle pixels will not match PCB drawing in some cases.

rhn, 2025-03-11:
I already used Lithophane to print 2 PCBs already - thanks, that wouldn't be possible without you.
I'll pay attention to the next exposure, but I think the Photon Mono SE is square.

sn4k3, 2025-03-11:
UVtools tell you the pixel pitch on bottom status bar, on Pixel: label. 
If square it will show only one value, if not it will show two values, eg: 14 x 19um