sn4k3/UVtools #1010: [Bug] UVTools v5.0.8 is unable to run

galmiklos, 2025-03-20:
### System

```Text
UVtools v5.0.8 X64
Operative system: Ubuntu 22.04.5 LTS X64
Processor: 12th Gen Intel(R) Core(TM) i7-12700KF
Processor cores: 20
Memory RAM: 62.02 / 62.63 GB
Runtime: linux-x64
Framework: .NET 9.0.2
AvaloniaUI: 11.2.5
OpenCV: 4.10.0

Sreens, resolution, working area, usable area:
1: 3440 x 1440 @ 100% (Primary) (On this)
    WA: 3440 x 1413    UA: 3440 x 1413

Path:       /tmp/.mount_UVtoolsZcNFM/usr/bin/
Executable: /tmp/.mount_UVtoolsZcNFM/usr/bin/UVtools
```

### Printer and Slicer

- Printer: N/A
- Slicer: N/A

### Description of the bug

I saw old issues about this library, it looks like the problem is back. I am trying to run UVtools on Ubuntu "22.04.5 LTS (Jammy Jellyfish)". I ran the dependency script.

```
gal@roki:~$ sudo bash -c "$(curl -fsSL https://raw.githubusercontent.com/sn4k3/UVtools/master/Scripts/install-dependencies.sh)"

Script to install the dependencies in order to run the UVtools
Hit:1 https://apt.grafana.com stable InRelease
Hit:2 https://download.docker.com/linux/ubuntu jammy InRelease                                                                                                                              
Hit:3 https://deb.nodesource.com/node_14.x jammy InRelease                                                                                                                                  
Hit:4 https://packages.microsoft.com/repos/code stable InRelease                                                                                                                            
Hit:5 https://dl.google.com/linux/chrome/deb stable InRelease                                                                                                                               
Hit:6 https://download.virtualbox.org/virtualbox/debian jammy InRelease                                                                                                                     
Hit:7 https://repo.mongodb.org/apt/ubuntu jammy/mongodb-org/7.0 InRelease                                                                                                                   
Hit:8 https://openhab.jfrog.io/artifactory/openhab-linuxpkg stable InRelease                                                                                                                
Hit:9 http://security.ubuntu.com/ubuntu jammy-security InRelease                                                                                                                            
Hit:10 http://us.archive.ubuntu.com/ubuntu jammy InRelease                                                                                                    
Hit:11 http://us.archive.ubuntu.com/ubuntu jammy-updates InRelease                        
Hit:12 http://us.archive.ubuntu.com/ubuntu jammy-backports InRelease                      
Hit:13 https://ppa.launchpadcontent.net/audio-recorder/ppa/ubuntu jammy InRelease         
Get:14 https://esm.ubuntu.com/apps/ubuntu jammy-apps-security InRelease [7,583 B]
Hit:15 https://ppa.launchpadcontent.net/freecad-maintainers/freecad-stable/ubuntu jammy InRelease
Get:16 https://esm.ubuntu.com/apps/ubuntu jammy-apps-updates InRelease [7,456 B]
Get:17 https://dl.packager.io/srv/deb/inventree/InvenTree/stable/ubuntu 20.04 InRelease [1,863 B]
Hit:18 https://ppa.launchpadcontent.net/giuspen/ppa/ubuntu jammy InRelease        
Get:19 https://esm.ubuntu.com/infra/ubuntu jammy-infra-security InRelease [7,450 B]
Hit:20 https://ppa.launchpadcontent.net/gns3/ppa/ubuntu jammy InRelease
Get:21 https://esm.ubuntu.com/infra/ubuntu jammy-infra-updates InRelease [7,449 B]
Hit:22 https://ppa.launchpadcontent.net/iconnor/zoneminder-1.36/ubuntu jammy InRelease
Hit:23 https://ppa.launchpadcontent.net/kicad/kicad-8.0-releases/ubuntu jammy InRelease
Fetched 31.8 kB in 1s (22.0 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
11 packages can be upgraded. Run 'apt list --upgradable' to see them.
- Detected: debian x86_64
Completed: You can now run UVtools
```

Then the install script.

```gal@roki:~/Applications$ bash -c "$(curl -fsSL https://raw.githubusercontent.com/sn4k3/UVtools/master/Scripts/install-uvtools.sh)"
  _   ___     ___              _     
 | | | \ \   / / |_ ___   ___ | |___ 
 | | | |\ \ / /| __/ _ \ / _ \| / __|
 | |_| | \ V / | || (_) | (_) | \__ \
  \___/   \_/   \__\___/ \___/|_|___/
  Auto download and installer script 

- Detecting OS
- linux x86_64
- Detecting download
- Downloading:  https://github.com/sn4k3/UVtools/releases/download/v5.0.8/UVtools_linux-x64_v5.0.8.AppImage
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 97.2M  100 97.2M    0     0  64.1M      0  0:00:01  0:00:01 --:--:--  110M
- Validating file
- Kill instances
- Removing old versions
- Moving UVtools_linux-x64_v5.0.8.AppImage to /home/gal/Applications
- Setting permissions
mode of '/home/gal/Applications/UVtools.AppImage' changed from 0600 (rw-------) to 0775 (rwxrwxr-x)

Installation was successful. UVtools will now run.
If prompt for "Desktop integration", click "Integrate and run"

```

And then, I get an error message.

[image]

### How to reproduce

1. Run dependency script.
2. Run install script
3. Error appears

sn4k3, 2025-03-20:
Upgrading OS to latest version an issue? 
The oficial emgucv doesn't support older systems, my custom compilation allowed that for a while, but we are sacrificing some combability to performance (Newer lib have better optimizations)

galmiklos, 2025-03-20:
Oh, I didn't realize 22 has already been 3 years ago, sorry. As a matter of fact, UVtools runs just fine on my "24.04.2 LTS (Noble Numbat)" machine. Yes, I can upgrade, no problem.

closed 2025-03-20 by galmiklos