sn4k3/UVtools #1013: [Bug] UVTools issue with Saturn 4 Ultra

henriquegdec, 2025-04-03:
### System

```Text
UVtools v5.0.5 X64
Operative system: Microsoft Windows 10.0.17763 X64
Processor: AMD Ryzen 5 5600 6-Core Processor              
Processor cores: 12
Memory RAM: 7.63 / 15.93 GB
Runtime: win-x64
Framework: .NET 9.0.0
AvaloniaUI: 11.2.3
OpenCV: 4.9.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1040    UA: 1920 x 1040
2: 1600 x 900 @ 100%
    WA: 1600 x 860    UA: 1600 x 860

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: 6 Exposure Tester 7 v2_1.goo [Version: 0] [Class: GooFile]
```

### Printer and Slicer

- Printer: Saturn 4 Ultra
- Slicer: Voxeldance Tango

### Description of the bug

I've had a Saturn 3 for a year and I've used UVTools on pretty much every single file I have printed and it always worked flawlessly, yesterday I bought a Saturn 4 Ultra and trouble started.
When I print a file without going through UV Tools, it works, but after removing islands, suction cups, setting exposures, etc...I get this:
https://imgur.com/a/BN1L1a9
(the normal exposure is that high because my first thought was underexposure)

I've tried to-reslice and even using a completely different model and the result was the same. After a bunch of failed attempts I decided to skip using UVTools and voila it worked, then I tried that first model without UVTools and it worked as well. 
So without any doubt it's happening because of UVTools, why specifically or which feature I'm not sure. This afternoon I'll try to just open the sliced file on UVTools and save it without modifying a single thing.
If it works, I'll try only removing the islands and so on until I find the exact cause

### How to reproduce

1. slice file for saturn 4 ultra
2. open on UVTools
3. do what UVTool does
4. print a mess

### Files

https://imgur.com/a/BN1L1a9

henriquegdec, 2025-04-03:
update:
Even without changing a single setting I got the same messed up result, the files:
https://drive.google.com/file/d/1IazA0UUobtkeHZ-yY8nGqaXZQxnuOSSM/view?usp=sharing

sn4k3, 2025-04-03:
I compared your files and the result is:

```
UVtoolsCmd compare "D:\Tiago\Desktop\cabecas dos desolators e o dripper_1_02h48m_78ml_copy.goo" "D:\Tiago\Desktop\cabecas dos desolators e o dripper_1.goo"
Opening file cabecas dos desolators e o dripper_1_02h48m_78ml_copy.goo: Done in 0.60s
Opening file cabecas dos desolators e o dripper_1.goo: Done in 0.03s
[Global #17]
BottomLiftHeight: 0.05 != 0
LiftHeight: 0.05 != 0
MaterialGrams: 77.422 != 85.085
MaterialCost: 15.484 != 0
BottomLiftHeightTotal: 0.05 != 0
LiftHeightTotal: 0.05 != 0
BottomRetractHeightTotal: 0.05 != 0
RetractHeightTotal: 0.05 != 0
BottomRetractHeight: 0.05 != 0
RetractHeight: 0.05 != 0
StartingMaterialCost: 15.484 != 0
```

UVtools force 0.05 (layer height) as Lift parameters as a rule for tilting VAT printer. Your printer use tilting vat correct?
Test the file, I now forced to 0, print and report back.

[cabecas dos desolators e o dripper_1_02h48m_78ml_uvtools0mm.zip](https://github.com/user-attachments/files/19592511/cabecas.dos.desolators.e.o.dripper_1_02h48m_78ml_uvtools0mm.zip)

henriquegdec, 2025-04-03:
> I compared your files and the result is:
>
> ```
> UVtoolsCmd compare "D:\Tiago\Desktop\cabecas dos desolators e o dripper_1_02h48m_78ml_copy.goo" "D:\Tiago\Desktop\cabecas dos desolators e o [...]

yep, that worked, so it's definitely an issue with one of those settings

sn4k3, 2025-04-03:
Ok, I will prepare a patch with 0s on those for tilting vat printer.

sn4k3, 2025-04-04:
You can test the new release

syskin345, 2025-04-12:
Are you saving the file as .goo or .ctb? I was never able to successfully print a UVTools-exported .goo file on S4U and simply stopped trying.

mindrush12, 2025-04-19:
I got the same issue! I'm just glad I found the cause and I'm not going crazy. I also have a Saturn 4 Ultra 16k.

[image]

mindrush12, 2025-04-19:
> Are you saving the file as .goo or .ctb? I was never able to successfully print a UVTools-exported .goo file on S4U and simply stopped trying. .ctb works perfectly.
> 
> To be precise, the printer prints a .goo file with layer height of 0 - the Z-axis just doesn't move.

So if I use a .ctb file on the S4U than it'll work fine?

syskin345, 2025-04-19:
> So if I use a .ctb file on the S4U than it'll work fine?

Well, it seems to always work fine for me, that's all I can promise.

Use `Convert To` -> `Chitubox` -> `Chitubox CTB (Encrypted)` -> `Version 5`

mindrush12, 2025-04-19:
> > So if I use a .ctb file on the S4U than it'll work fine?
> 
> Well, it seems to always work fine for me, that's all I can promise.
> 
> Use `Convert To` -> `Chitubox` -> `Chitubox CTB (Encrypted)` -> `Version 5`

Haha I'm afraid to try it. I wish there was another way to fill voids! It seems the models from titancraft always have voids.

sn4k3, 2025-04-19:
> I got the same issue! I'm just glad I found the cause and I'm not going crazy. I also have a Saturn 4 Ultra 16k.

You saying that still have the issue with recent UVtools? That should be fixed...
Send me your problematic file.

mindrush12, 2025-04-19:
> > I got the same issue! I'm just glad I found the cause and I'm not going crazy. I also have a Saturn 4 Ultra 16k.
> 
> You saying that still have the issue with recent UVtools? That should be fixed... Send me your problematic file.

Attached is the file I printed that failed.

sn4k3, 2025-04-19:
> It won't let me attach the file.

Compress to zip, if still not work use any file transfer host, like wetransfer.com

mindrush12, 2025-04-19:
Try this https://we.tl/t-bey2OnUT0b

sn4k3, 2025-04-19:
> Try this https://we.tl/t-bey2OnUT0b

That file was not processed with UVtools. I need one that was and is failing.

mindrush12, 2025-04-19:
Hmm I thought it was. Try this one https://we.tl/t-GYat1wJlBx

mindrush12, 2025-04-19:
You know what it could of been the USB. Here is the file from my USB. https://we.tl/t-w43YfaXK2d

sn4k3, 2025-04-19:
The problem on your case looks related to using per layer settings, it's active due a precision error on layer height which UVtools turns on when found difference between globals and layer settings.
I will include a tweak to handle this precision errors making the flag off in this case.

To solve your situation now, you can go to `File and while hold SHIFT key, press on Reset layer properties`

mindrush12, 2025-04-20:
Thank you! How are you able to see this information in the files? Before I go test the print, can you look over these files? https://we.tl/t-xzJD5O8PGj

uv fixed goo file no reset: this file is the baseline without reseting layer properties
uv fixed goo file with layer options reset: this is a goo file with the reset layer properties done
uv fixed ctb file: I made a ctb file as well which was reccomended.

sn4k3, 2025-04-20:
To make sure file is safe to print, I recommend to look in UVtools information panel, look for this flag, which should be `False`:

[image]

Note that flag is only changed when you save the file. So always check after save file.

mindrush12, 2025-04-20:
Alright I am a bit more confused now. I tested looking for the False condition with two files. The file I ran through UVtools and didn't follow your advice to "go to File menu and while hold SHIFT key, press on Reset layer properties", it has the False condition which you said it should be. The file where I did "reset layer properties", I got this error:

[image]

sn4k3, 2025-04-20:
Strange thing, can you get a way to reproduce this and send me the steps?

mindrush12, 2025-04-20:
Yes I reproduced. 

1. Open UVtools
2. open .goo file
3. Click "Issues"
4. Click "Fix Issues"
5. Select all "Resin Trap" issues
6. Right click and fill traps
7. Save
8. Close UVtools
9. Open UVtools
10. open new saved .goo file.

Now when I just use default settings and don't shift click the reset layer properties in the file open menu it opens fine and it defaults to FALSE on PerLayerSettings. I just had a successful test print ensuring the file said False under PerLayerSettings. So I'm not sure what my initial problem was. Was it the PerLayerSettings = 0?

betraymydream, 2025-04-20:
I may I found a solution to your problem. Check it out here [click](https://forreststonesolutions.com/robots/)

sn4k3, 2025-04-21:
The shift key is only required if your file have rounding problems on layer height. Otherwise you can run without shift.
PerLayerSettings = 0 = false. But no error you post is not related to that.

mindrush12, 2025-04-21:
Alright thank you. My most recent print was successful. It doesn't feel good that everytime I use UVtools its like a gamble lol. Is there any way to guarantee it won't happen? The flattening of my models is what I'm talking about. Last one caused a hole in my FEP screen.

sn4k3, 2025-04-22:
Check latest release

mindrush12, 2025-04-22:
I'll check it out! Should I always use CTB format? What is the difference from the elegoo one?

sn4k3, 2025-04-22:
Goo format should be better in papper, but ctb have the implementation better.
But in reality no format for mSLA is good.

sn4k3, 2025-04-25:
It must have any discrepancy with the layers again.
You can send the file directly to my email: Tiago_caza@hotmail.com

jwcrawley, 2025-06-02:
Id like to add the same problem. 

--------------------
UVtools v5.1.2 X64
Operative system: Ubuntu 24.04.2 LTS X64
Processor: AMD Ryzen 7 7700X 8-Core Processor
Processor cores: 16
Memory RAM: 41.56 / 93.39 GB
Runtime: linux-x64
Framework: .NET 9.0.5
AvaloniaUI: 11.3.0
OpenCV: 4.10.0

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 100% (Primary) (On this)
    WA: 2560 x 1415    UA: 2560 x 1415

Path:       /tmp/.mount_UVtool1B0ouY/usr/bin/
Executable: /tmp/.mount_UVtool1B0ouY/usr/bin/UVtools
Loaded file: garklein.stl_0.05_1.2_2025_06_02_12_57_00.goo [Version: 0] [Class: GooFile]

_________________

I bought a Saturn 4 Ultra 2 days ago. I use Chitubox Basic, and UVTools to validate and get rid of the 1-3 pixel islands Chitu does, using UVTools. 

When I did this 2x, I ended up with the support raft for the part, but none of the actual part, as shown with bug reports. Picture shown.

[image]

This is an instrument I made myself, called a Garklein. I retain copyright, and intend to offer under a permissive license. I provide the files here for troubleshooting purposes. 

Goo and stl files both provided.

[garklein.stl_0.05_1.2_2025_06_02_12_57_00.goo.zip](https://github.com/user-attachments/files/20557993/garklein.stl_0.05_1.2_2025_06_02_12_57_00.goo.zip)

[garklein.stl.zip](https://github.com/user-attachments/files/20557996/garklein.stl.zip)

jwcrawley, 2025-06-02:
I switched to the CTB format instead. And its only at 2% done, but the Z bed does appear to be progressing as it should.

I can see small (.05mm) progressive increases of the leadscrew.

The part is still under liquid, but the issue of Z not incrementing at least is triaged with .ctb file format.